seemoo-lab/nexmon #379: Nexus 5 with Nexmon, but "airodump-ng wlan0" no results

Sma11M00n, 2020-02-07:
hi,
i got a nexus5 with:
- android 6.0.1 (M4B30Z, Dec 2016) from https://developers.google.com/android/images
- twrp 3.3.1
- SuperSU v2.82
- nethunter-hammerhead-marshmallow-kalifs-full-2020.1

Then i built nexmon followed by the procedure "Build patches for bcm4330, bcm4339 and bcm4358 using a x86 computer running Linux"

everything is ok but when i use kali-terminal with "LD_PRELOAD=libfakeioctl.so airodump-ng wlan0", it shows:
`ERROR: 
ld.so: object 'libfakeioctl.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ioctl(SIOCSIWMODE) failed: Operation not supported
ioctl(SIOCSIWMODE) failed: Operation not supported
Error setting monitor mode on wlan0
Failed initializing wireless card(s): wlan0
`

Then, i try same command in adb-terminal with root, "LD_PRELOAD=libfakeioctl.so airodump-ng wlan0" is success **but no results and no errors.**
libfakeioctl.so is in /system/lib and airodump-ng is in /system/bin, both them are built in nexmon/utilities

Also, i try Hijack app in my devices, and it's also the same situation.

So, What I do wrong ?

P.S. maybe i need build nexmon in nethunter, but i can't find any guide，please help me with this. Thanks a lot

baselsayeh, 2020-02-18:
(cannot open shared object file) means that you did not install the library file properly.
See [this page](https://github.com/seemoo-lab/nexmon/blob/master/README.md#using-the-monitor-mode-patch) for how to use the patch correctly.
Also, you may need to compile the utilities manually because the nexmon app is outdated, see [This page](https://github.com/seemoo-lab/nexmon/blob/master/README.md#using-the-monitor-mode-patch) for how to compile the utilities
Basically:
download and extract ndk 11c
export NDK_ROOT=/path/to/ndk11c
cd utilities
make

YaohuaLee, 2020-06-08:
@baselsayeh 
sir, I had a similar problem make the nexutil
```
In file included from ./nexutil.c:39:0:
./../libargp/argp.h:583:3: error: #error "Please include config.h first."
  #error "Please include config.h first."
   ^
./../libargp/argp.h:585:1: error: unknown type name '_GL_INLINE_HEADER_BEGIN'
 _GL_INLINE_HEADER_BEGIN
 ^
./../libargp/argp.h:616:9: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 ARGP_EI void
         ^
./../libargp/argp.h:622:9: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'int'
 ARGP_EI int
         ^
./../libargp/argp.h:634:9: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'int'
 ARGP_EI int
         ^
./../libargp/argp.h:645:1: error: unknown type name '_GL_INLINE_HEADER_END'
 _GL_INLINE_HEADER_END
 ^
In file included from /home/yxy/android-ndk-r11c/platforms/android-21/arch-arm64/usr/include/byteswap.h:31:0,
                 from ./nexutil.c:41:
/home/yxy/android-ndk-r11c/platforms/android-21/arch-arm64/usr/include/sys/endian.h:52:10: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'htonl'
 uint32_t htonl(uint32_t) __pure2;
          ^
In file included from /home/yxy/nexmon/patches/include/asm/socket.h:19:0,
                 from ./../../patches/include/linux/socket.h:27,
                 from /home/yxy/android-ndk-r11c/platforms/android-21/arch-arm64/usr/include/sys/socket.h:34,
                 from ./nexutil.c:44:
/home/yxy/nexmon/patches/include/asm-generic/socket.h:21:65: fatal error: home/yxy/nexmon/patches/include/asm-generic/sockios.h: No such file or directory
 #include <home/yxy/nexmon/patches/include/asm-generic/sockios.h>
                                                                 ^
compilation terminated.
/home/yxy/android-ndk-r11c/build/core/build-binary.mk:460: recipe for target 'local/arm64-v8a/objs/nexutil/nexutil.o' failed
make[1]: *** [local/arm64-v8a/objs/nexutil/nexutil.o] Error 1
make[1]: Leaving directory '/home/yxy/nexmon/utilities/nexutil'
Makefile:11: recipe for target 'libs/armeabi/nexutil' failed
make: *** [libs/armeabi/nexutil] Error 2
```
Here are some of my environment

```
export NDK_ROOT=/home/yxy/android-ndk-r11c/
export PATH=$NDK_ROOT:$PATH
```
Can you give me some help？