seemoo-lab/nexmon #381: Best capability WiFi module for new device

zhovner, 2020-02-10:
We are developing a portable device for security researchers in a Tamagotchi-style formfactor called [Flipper Zero](https://flipperzero.one/). We decide to drop Raspberry Pi and build our board from scratch and now looking for a WiFi module that supports all features for wifi hacking.

If Seemoo guys don't mind I will post my research of Broadcom modules here. Any comments are welcome.

### Our requirements

* SDIO 2.0/3.0 interface
* Dual-band (2.4 and 5 GHz) 802.11ac on a single antenna
* System in Package (SiP) module — this modules already have all RF components like LNA, filters, etc packaged in one tiny PCB covered with metal shield.
* Monitor mode
* Packet injection

## Possible candidates
I will post all my finds here and update this post while testing. Will open every module to see what the chipset inside. If you know some candidates, please suggest it in the same style.

## Ampak AP6255
Chipset: `BCM43454HKUBG`
802.11ac, bluetooth 4.2
Installed on Orange Pi Lite 2. Internet tells that this is bcm43455c0 but i’m not sure and can't figure out what the difference between `bcm43455` and _bcm43455_**C0**. Raspberry Pi 3B+ have `CYW43455XKUBG` chipset. 
On Orange Pi Lite 2 with Armbian distro it use the same firmware as Raspbian — `7.45.154`.
[image]

```
$ dmesg | grep brc

bluetooth hci1: Direct firmware load for brcm/BCM4345C0.hcd failed with error -2
Bluetooth: hci1: BCM: Patch brcm/BCM4345C0.hcd not found
brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.xunlong,orangepi-lite2.txt failed with error -2
brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04

$ grep "" /sys/class/mmc_host/mmc1/mmc1\:0001/mmc1\:0001\:*/{class,device,vendor}
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/class:0x00
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/class:0x00
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:3/class:0x02
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/device:0xa9bf
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/device:0xa9bf
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:3/device:0xa9bf
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/vendor:0x02d0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/vendor:0x02d0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:3/vendor:0x02d0

$ cat /sys/kernel/debug/brcmfmac/mmc1\:0001\:1/revinfo
vendorid: 0x14e4
deviceid: 0x43ab
radiorev: 0.88.3.11
chipnum: 17221 (4345)
chiprev: 6
chippkg: 2
corerev: 54
boardid: 0x06e4
boardvendor: 0x14e4
boardrev: P304
driverrev: 7.45.18
ucoderev: 0
bus: 0
phytype: 11
phyrev: 20
anarev: 0
nvramrev: 00079ac5
```

## Ampak AP6256
Chipset: `BCM43456XKUBG`
802.11ac, bluetooth 5
Installed on Orange Pi 3. Not listed in nexmon supported hardware. Maybe can be suitable.  
Uses this firmware [brcmfmac4356-sdio.bin](https://github.com/armbian/firmware/blob/master/brcm/brcmfmac4356-sdio.bin)
[image]

```
# dmesg | grep brcm
brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
brcmfmac mmc0:0001:1: Direct firmware load for brcm/brcmfmac43456-sdio.xunlong,orangepi-3.txt failed with error -2
brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/9 wl0: Jun 16 2017 12:38:26 version 7.45.96.2 (66c4e21@sh-git) (r) FWID 01-1813af84

 /sys/bus/sdio/devices/mmc0:0001:1/vendor:0x02d0
 /sys/bus/sdio/devices/mmc0:0001:1/device:0xa9bf

```

Icenowy, 2020-02-12:
I assume 45455c0 start to be called c0 when c5 is available.

jorikdima, 2020-02-27:
In Broadcom the last letter means chip tapeout revision and the last digit - ECO (ROM update e.g.). In this case part number without these Character&Digit means it's just omitted. Most likely it has B? in the end. Switching from B tapeout revision to C means that some minor HW was probably added or fixed, but in general it's still the same chip. Requirements to use only C* firmware means this FW uses this new HW.