seemoo-lab/nexmon #384: Monitor mode on raspberry pi 3b+

Impeekay, 2020-02-23:
Monitor mode on 3b+ with the nexmon patch seems to be crashing every now and then, and the only solution to it seems to be rebooting the device, I've also looked at the issue in raspberry pi itself, but I'm hoping this might an issue with the patch also.

after some time running it on monitor mode, this is what I get on dmesg and it stops responding
```
A LOT OF THESE
[  749.827012] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4108, -52
[  751.968287] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4109, -52
[  754.107140] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4110, -52
[  772.533950] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: Enter
[  772.533966] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: 0058454e 20 36
[  772.533976] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: calling brcmf_fil_cmd_data_get, cmd: 107
[  772.534536] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: Exit
[  775.884183] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: Enter
[  775.884200] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: 0058454e 20 36
[  775.884210] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: calling brcmf_fil_cmd_data_get, cmd: 107
[  775.884763] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: Exit
[  779.776599] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4108, -52
[  781.917153] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4109, -52
[  784.052354] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4110, -52
[  809.705557] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4108, -52
A LOT OF THESE AGAIN
[ 810.7987667] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmare halted
[ 810.7989997] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[ 810.7800199] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[ 810.7819922] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53292, -110

AND AGAIN A LOT OF THESE TILL REBOOT
```

zeroby0, 2020-02-24:
@Impeekay what power supply are you using? I've had issues with bad power supplies and power banks.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Impeekay, 2020-02-25:
@zeroby0 no i did check the power supply, I've plugged it directly to the power. Also the issue is it runs for certain hour's and then stops.

DrSchottky, 2020-02-25:
#280

Impeekay, 2020-02-26:
@DrSchottky, thanks for your reply, yes I did check that issue before raising an issue. I'm sure and I have confirmed that I have the correct patches built and installed. Hence, raised this issue here.

DrSchottky, 2020-02-26:
@Impeekay We need to know more about the context:
What are you doing? Only monitoring or also injection? With wich sw?
What's in your `forensics` file? Kernel and FW version?
And so on...

Impeekay, 2020-02-26:
@DrSchottky , that is the thing that is bothering me, the `forensics` file does not exist in mine at all.