seemoo-lab/nexmon #387: Cannot build for Nexus 6P nougat 7.1.2

OstojaOfficial, 2020-03-11:
l am building this on Kali Linux x64 2020.1
I correctly setup NDK and cloned the Nexmon from GitHub but when I try to build it for nougat 7.1.2 (7_112_201_3_sta) It fails with error message:
`
COMPILING src/ioctl.c => obj/ioctl.o (details: log/compiler.log)
src/ioctl.c:48:21: fatal error: bcmpcie.h: No such file or directory
compilation terminated
make: *** [Makefile:42: obj/ioctl.o] Error 1
`
log/compiler.log is empty but in log/error.log I have this:
`
cat: gen/nexmon.pre: No such file or directory
`

I have no idea how to fix this issue.

sylar1233, 2020-03-12:
three years ago it was true that nexmon only supported oreo.
Maybe it's because you try nougat.
as far as it hasn't changed ...

TheMMcOfficial, 2020-03-13:
I suggest you to get this version of android:
angler-opr6.170623.019-factory-9fd72ad6 (8.0.0 (OPR6.170623.019, Sep 2017))

Or you will need to get the kernel of @kimocoder
https://github.com/chrisk44/Hijacker/issues/53

I will link you some bug that I encounter during my test: https://github.com/seemoo-lab/nexmon/issues/195

OstojaOfficial, 2020-03-14:
> 
> 
> I suggest you to get this version of android:
> angler-opr6.170623.019-factory-9fd72ad6 (8.0.0 (OPR6.170623.019, Sep 2017))
> 
> Or you will need to get the kernel of @kimocoder
> [chrisk44/Hijacker#53](https://github.com/chrisk44/Hijacker/issues/53)
> 
> I will link you some bug that I encounter during my test: #195

On README.md of this repo it says under supported devices that Nexmon Firmware version 7_112_201_3_sta supports Android version 7.1.2, I would run 8.0.0 but I have NetHunter for my Nexus 6P. On NetHunter repo it says that anything above 7.1.2 can cause USB problems.

Wiki page on NetHunter repo: 
[image]
https://gitlab.com/kalilinux/nethunter/build-scripts/kali-nethunter-project/-/wikis/home

README.md for this repo:
[image]
https://github.com/seemoo-lab/nexmon

TheMMcOfficial, 2020-03-14:
Do you build the nexmon firmware on a debian base Linux system ? 
Do you have all the requirements ?

```

    Install some dependencies: sudo apt-get install git gawk qpdf adb flex bison

    Only necessary for x86_64 systems, install i386 libs:

    sudo dpkg --add-architecture i386
    sudo apt-get update
    sudo apt-get install libc6:i386 libncurses5:i386 libstdc++6:i386

    Clone our repository: git clone https://github.com/seemoo-lab/nexmon.git

    In the root directory of the repository: cd nexmon
        Setup the build environment: source setup_env.sh
        Compile some build tools and extract the ucode and flashpatches from the original firmware files: make

    Go to the patches folder of your target device (e.g. bcm4339 for the Nexus 5): cd patches/bcm4339/6_37_34_43/nexmon/
        Compile a patched firmware: make
        Generate a backup of your original firmware file: make backup-firmware
        Install the patched firmware on your smartphone: make install-firmware (make sure your smartphone is connected to your machine beforehand)

Using the Monitor Mode patch

    Install at least nexutil and libfakeioctl from our utilities. The easiest way to do this is by using this app: https://nexmon.org/app. But you can also build it from the source by executing make in the utilties folder (Note: you will need the Android NDK properly installed for this).
    Connect to your Android phone using the ADB tools: adb shell
    Make sure you are not connected to an access point
    Use nexutil to enable monitor mode: nexutil -m2
    At this point the monitor mode is active. There is no need to call airmon-ng.
    Important: Most tools need a Radiotap interface to work properly. libfakeioctl emulates this type of interface for you, therefore, use LD_PRELOAD to load this library when you call the favourite tool (e.g. tcpdump or airodump-ng): LD_PRELOAD=libfakeioctl.so tcpdump -i wlan0
    untested hint: Thanks to XDA member ruleh, there is a bcmdhd driver patch to activate native monitor mode, see: https://github.com/ruleh/misc/tree/master/monitor

```

OstojaOfficial, 2020-03-16:
> Do you build the nexmon firmware on a debian base Linux system ?
> Do you have all the requirements ?
>
> ```
>
>     Install some dependencies: sudo apt-get install git gawk qpdf adb flex bison
>
>     Only [...]

Yes, I have all the requirements, I am using Kali Linux which should be Debian base Linux system and can build that one but I can't build this one (patches/bcm4358/7_112_201_3_sta). Can you try to build that as well? I have seen another person also made an isshe regarding this same thing.

TheMMcOfficial, 2020-03-16:
Do you have the NDK install I compiled them some months/year ago I don't remember if I got trouble to compile them...

sylar1233, 2020-03-16:
can you please give me these simple answers thank you
https://github.com/seemoo-lab/nexmon/issues/387

OstojaOfficial, 2020-03-17:
> Do you have the NDK install I compiled them some months/year ago I don't remember everything I did but I think it compiled fine.

Yes I have NDK r11c installed and in my PATH.