seemoo-lab/nexmon #388: Helo

sylar1233, 2020-03-12:
my device nexus 6p
I apologize in advance for my english. It's been 3 years since I've been dealing with kali nethunter and nexmon.
I want to deal with it again and I have a few questions about it.
Which rome and which kernel is best for kali and nexmon?
And which nethunter build or chroot build do you recommend?
Three years ago it was only possible to use it with stock rom nexmon if I remember correctly.
thank you for your coming help.

TheMMcOfficial, 2020-03-25:
For nexmon I suggest you this android version: angler-opr6.170623.019-factory-9fd72ad6 (8.0.0 (OPR6.170623.019, Sep 2017))
For nethunter is kind of difficult question because I don't think they support the oreo version I need to recheck to be sure but they had bugs with the hid keyboard add to the kernel.

They don't have this bug on android 7 but the kernel needed some change to be able to use the injection (I think Kimcoder had done it please read this issue https://github.com/chrisk44/Hijacker/issues/53#issuecomment-554683457). I'm not an expert I just read I lot of github and gitlab issue to be able to make my nexus 6p working with nethunter.

kimocoder, 2020-03-25:
@TheMMcOfficial We're building a new angler kernel over at NetHunter.

@re4son is on the case, find repos below
https://github.com/Re4son?tab=repositories

Find us at Telegram
https://t.me/joinchat/AAAAAFDVPDIHabBJwhL1Mw

TheMMcOfficial, 2020-03-26:
Thanks for your work @kimocoder and https://github.com/Re4son

kimocoder, 2020-03-26:
@TheMMcOfficial No problem, thanks for yours too upon these years. And NetHunter supports up to Android 10 at this point :+1: 

Will reveal a patch for Qualcomm QCACLD driver  tomorrow, activate monitor mode on phones using the driver :1st_place_medal:

TheMMcOfficial, 2020-05-23:
@kimkoder I have try the new version of nethunter 2020.2 but I don't think I able to put my internal wireless card in monitor mode is it normal ?

I build my nethunter on top of PixelExperience_angler-10.0-20200101-0925... like the XDA post suggest.