seemoo-lab/nexmon #392: Samsung Galaxy S8+ (G995F)

KuLiD, 2020-04-09:
Hi
Awsome work Guys!!
Is the S8 firmware patch (bcm4361b0) compatible with the S8+ firmware (bcm4316b0_murata)?

Thank You

matthiasseemoo, 2020-04-09:
try it.

KuLiD, 2020-04-09:
I guess :)

KuLiD, 2020-04-09:
@matthiasseemoo 
[armeabi-v7a] Compile thumb  : nexutil <= nexutil.c
[armeabi-v7a] Executable     : nexutil
./argp-help.c:1643: error: undefined reference to 'argp_fmtstream_set_wmargin'
./argp-help.c:1657: error: undefined reference to 'argp_fmtstream_set_lmargin'
./argp-help.c:1663: error: undefined reference to 'argp_fmtstream_puts'
./argp-help.c:1675: error: undefined reference to 'argp_fmtstream_set_wmargin'
./argp-help.c:1676: error: undefined reference to 'argp_fmtstream_set_lmargin'
./argp-help.c:1678: error: undefined reference to 'argp_fmtstream_putc'
./argp-help.c:1704: error: undefined reference to 'argp_fmtstream_putc'
./argp-help.c:1111: error: undefined reference to 'argp_fmtstream_set_lmargin'
./argp-help.c:1132: error: undefined reference to 'argp_fmtstream_set_wmargin'
./argp-help.c:1140: error: undefined reference to 'argp_fmtstream_putc'
./argp-help.c:1141: error: undefined reference to 'argp_fmtstream_putc'
./argp-help.c:1174: error: undefined reference to 'argp_fmtstream_set_wmargin'
./argp-help.c:1162: error: undefined reference to 'argp_fmtstream_puts'
./argp-help.c:1189: error: undefined reference to 'argp_fmtstream_set_lmargin'
./argp-help.c:1208: error: undefined reference to 'argp_fmtstream_point'
./argp-help.c:1216: error: undefined reference to 'argp_fmtstream_puts'
./argp-help.c:1220: error: undefined reference to 'argp_fmtstream_puts'
./argp-help.c:1460: error: undefined reference to 'argp_fmtstream_write'
./argp-help.c:1550: error: undefined reference to 'argp_fmtstream_point'
./argp-help.c:1572: error: undefined reference to 'argp_fmtstream_point'
./argp-help.c:956: error: undefined reference to 'argp_fmtstream_point'
collect2: error: ld returned 1 exit status
make[2]: *** [/opt/android-ndk-r11c/build/core/build-binary.mk:679: local/armeabi-v7a/nexutil] Error 1
make[2]: Leaving directory '/home/dalynn/nexmon/utilities/nexutil'
make[1]: *** [Makefile:12: libs/armeabi/nexutil] Error 2
make[1]: Leaving directory '/home/dalynn/nexmon/utilities/nexutil'
make[1]: Entering directory '/home/dalynn/nexmon/utilities/netcat'

I have followed every step. Installed NDK-R11C, Set NDK_ROOT but im getting this now.
Any help would be appreciated:)

When I do MAKE INSTALL everything shows except nexutil on my mobile.

KuLiD, 2020-04-12:
@matthiasseemoo 
S8+
TWRP
MAGISK
NETHUNTER
I managed to manually transfer all the nexmon libs etc, I then installed the firmware via Kali Linux. I now have a error:
nexutil -m
_nex_driver_io:error
Monitor 0

04-12 15:01:07.445  3232  3232 E audit   : type=1400 audit(1586696467.439:399): avc:  denied  { ioctl } for  pid=18406 comm="nexutil" path="socket:[157345]" dev="sockfs" ino=157345 ioctlcmd=89f0 scontext=u:r:magisk:s0 tcontext=u:r:magisk:s0 tclass=udp_socket permissive=0 SEPF_SM-G955F_9_0013 audit_filtered

matthiasseemoo, 2020-04-12:
what is your wifi interface name? did you execute nexutil as root? is selinux set to permissive?

KuLiD, 2020-04-14:
@matthiasseemoo 
My wifi interface name is wlan0

I tried to excute via Nethunter because SElinux is disabled(getenforce) but if i use "getenforce" on AndroidSU terminal its enforcing. I have been trying to get it to permissive but its locked down tight, seems the only way is to get an Open source kernel for the G99F and disabled it in the script and then build it but if you know any other way that would be great. 

Another problem i picked up was that the firmware for the b4361b0 isnt compiling or extracting. It shows that its busy when executing but the hash sums are exactly the same on mobile and git repo...so im not sure whats happening.