seemoo-lab/nexmon #394: External antenna, estimated AOA

289536718, 2020-04-14:
Hello. 
I'm trying to use Raspberry 4B to estimate AOA from CSI, but it needs an external antenna. So how do I do it? Can you give me some advice?

Thank you very much.

mzakharo, 2020-04-14:
pardon for my ignorance, but how do you obtain AOA from just 1 RX antenna?

289536718, 2020-04-15:
Thank you for your reply. 
I learned about the “Nexmon”  a few days ago. When I asked this question, I didn't have the actual Raspberry 3B+/4B. I even thought that the network card of Raspberry  would have an interface exposed (like the Inter 5300). 
Later I found that it had no interface, but I wanted to use the CSI collected by Raspberry 4B to meet the estimation of AOA, so what should I do? Can it be realized through external network card?
Please give me some suggestions.

289536718, 2020-04-15:
> 请原谅我的无知，但是如何从1个RX天线获得AOA？
Thank you for your reply.
I learned about the “Nexmon” a few DayS ago. When I asked this question, I didn't have the actual Raspberry 3B+/4B. I even thought that the network card of Raspberry would have an interface exposed (like the Inter 5300).
Later I found that it had no interface, but I wanted to use the CSI collected by Raspberry 4B to meet the estimation of AOA,.so what should I do?
Can it be realized through external network card? Besides, does Raspberry's network card only have one antenna?
Please give me some suggestions.
Thank you very much.

289536718, 2020-04-15:
@mzakharo 
Thank you for your reply.
I learned about the “Nexmon” a few DayS ago. When I asked this question, I didn't have the actual Raspberry 3B+/4B. I even thought that the network card of Raspberry would have an interface exposed (like the Inter 5300).
Later I found that it had no interface, but I wanted to use the CSI collected by Raspberry 4B to meet the estimation of AOA,.so what should I do?
Can it be realized through external network card? Besides, does Raspberry's network card only have one antenna?
Please give me some suggestions.
Thank you very much.

matthiasseemoo, 2020-04-15:
If you want to collect CSI on multiple antennas, use one of the multi-antenna smartphones or wi-fi access points supported by nexmon. If you want something cheaper than the access point, you can also go for PCIE extension cards that has a supported wi-fi chip.

289536718, 2020-04-16:
@matthiasseemoo 
Mr.Matthias，Thank you for your reply.
Your answer has helped me a lot. Now I'm going to make an AP with Raspberry 4B and use ASUS rt-ac86u as receiver to collect  CSI for AOA estimation on 80 MHz .Can I do that?

closed 2020-04-26 by 289536718

tweigel-dev, 2020-12-01:
@289536718 Did you get CSI per annenna at the ASUS RtAC86u?  Can you please publish a pcap of such a collection?