seemoo-lab/nexmon #396: tracker

e2002e, 2020-04-29:
Why is there a google "analytics" in there ? did the guys at offensive security add it themselves ? my version comes from the nethunter store. We know where it's going, first google get all the datas, then the nsa or whatever agency hacks them...

matthiasseemoo, 2020-04-29:
A couple of years ago, one of my students built the app as part of his master thesis and also performed some analysis on the usage of the tool using google analytics and a survey. After a while, I wanted to get rid of the analysis code but never found the time. Instead of changing the app, I just removed it from the Play Store and kept the code for reference in this repo.

I would suggest the following: if you find some time to get rid of the tracking code, feel free to edit the app and submit your changes as a pull request. I can then merge it to the repo and a new version of the app can be released without tracking.

e2002e, 2020-04-29:
I would gladly but I can't seem to locate the parts of the code that make the android package
Is there a github command to get them ?
I succesfully installed the firmware 4330 for the nexus 7 and now am building the tools, I hope it will work. Your readme is very good.

matthiasseemoo, 2020-04-29:
Here is an example commit, but there should be more.
https://github.com/seemoo-lab/nexmon/commit/f0a654128cc88386b3e55a5b43a452ba0babbf64 <https://github.com/seemoo-lab/nexmon/commit/f0a654128cc88386b3e55a5b43a452ba0babbf64>

e2002e, 2020-04-29:
oh i see, i neglected the 'app' folder. I'll be looking into that soon. Thanks for your replies.

e2002e, 2020-04-29:
I am getting undefined references to argp_fmtstream_* when building the utilities.
This occurs when building armeabi-v7a's librairies, previous librairies compile well. Do you have a clue ? I picked the ndk indicated in the readme. firmware build well.

....
[mips64] Compile        : nexutil <= nexutil.c
[mips64] Executable     : nexutil
[mips64] Install        : nexutil => libs/mips64/nexutil
[armeabi-v7a] Compile thumb  : nexutil <= nexutil.c
[armeabi-v7a] Executable     : nexutil
./argp-help.c:1643: error: undefined reference to 'argp_fmtstream_set_wmargin'
./argp-help.c:1657: error: undefined reference to 'argp_fmtstream_set_lmargin'
./argp-help.c:1663: error: undefined reference to 'argp_fmtstream_puts'
....
make[2]: *** [/home/user/android-ndk-r11c/build/core/build-binary.mk:679: local/armeabi-v7a/nexutil] Error 1

closed 2020-05-05 by e2002e