seemoo-lab/nexmon #397: Nexmon crashes on fresh install on Kali Nethunter Nexus 5

2fst4u, 2020-04-30:
I'm not sure if I'm doing something wrong but this is a completely fresh install on Kali Nethunter on a Nexus 5 (which I've done several times before and is otherwise working).

When I install Nexmon from the Kali store version and attempt the "start" phase on first opening, root is granted and then the app just crashes giving the error "Unfortunately, Nexmon has stopped". After opening a second time (in case the lack of initial root was the issue) it does it again and every time after.

Should I be using a different version?

hva314, 2020-05-06:
I'm having the same issue on my Nexus 5. Here's a bit more context for my case.

Device model: Nexus 5, D821
Android version: 6.0.1
CyanogenMod version: 13.0
Magisk and Nethunter 2020.2-rc3 installed

adb logcat when the crash happen:
`05-06 17:20:10.052   857   872 I ActivityManager: Waited long enough for: ServiceRecord{db3fb8e u0 de.tu_darmstadt.seemoo.nexmon/.net.RawSocketReceiveService}
05-06 17:20:23.748   857   975 D AudioService: Stream muted, skip playback
05-06 17:20:23.925 30549 30549 D AndroidRuntime: Shutting down VM
05-06 17:20:23.926 30549 30549 E AndroidRuntime: FATAL EXCEPTION: main
05-06 17:20:23.926 30549 30549 E AndroidRuntime: Process: de.tu_darmstadt.seemoo.nexmon, PID: 30549
05-06 17:20:23.926 30549 30549 E AndroidRuntime: Theme: themes:{default=overlay:org.cyanogenmod.hexolibre, iconPack:org.cyanogenmod.hexolibre, fontPkg:org.cyanogenmod.hexolibre, com.android.systemui=overlay:org.cyanogenmod.hexolibre, com.android.systemui.navbar=overlay:org.cyanogenmod.hexolibre}
05-06 17:20:23.926 30549 30549 E AndroidRuntime: java.nio.BufferUnderflowException
05-06 17:20:23.926 30549 30549 E AndroidRuntime: 	at java.nio.ByteArrayBuffer.getInt(ByteArrayBuffer.java:196)
05-06 17:20:23.926 30549 30549 E AndroidRuntime: 	at de.tu_darmstadt.seemoo.nexmon.utils.Nexutil.getIntIoctl(Nexutil.java:194)
05-06 17:20:23.926 30549 30549 E AndroidRuntime: 	at de.tu_darmstadt.seemoo.nexmon.utils.FirmwareUtil.isBroadcomChip(FirmwareUtil.java:27)
05-06 17:20:23.926 30549 30549 E AndroidRuntime: 	at de.tu_darmstadt.seemoo.nexmon.gui.StartFragment$1.onClick(StartFragment.java:123)
05-06 17:20:23.926 30549 30549 E AndroidRuntime: 	at android.view.View.performClick(View.java:5204)
05-06 17:20:23.926 30549 30549 E AndroidRuntime: 	at android.view.View$PerformClick.run(View.java:21158)
05-06 17:20:23.926 30549 30549 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:739)
05-06 17:20:23.926 30549 30549 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:95)
05-06 17:20:23.926 30549 30549 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:148)
05-06 17:20:23.926 30549 30549 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:5461)
05-06 17:20:23.926 30549 30549 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
05-06 17:20:23.926 30549 30549 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
05-06 17:20:23.926 30549 30549 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
05-06 17:20:23.927   857  1801 W ActivityManager:   Force finishing activity de.tu_darmstadt.seemoo.nexmon/.gui.MyActivity
`

ctrl-freak, 2020-05-08:
I too am having this problem; though on a different device and OS.

Device: Nexus 7 2012 (tilapia)
Unlegacy Android 7.1.2
SuperSU and Nethunter 2020.2-rc3
Kernel: 3.4.113

Nexmon crashes when pressing the 'GRANT ROOT, PERMISSIONS & CHECK INSTALLATION' button and the same BufferUnderflowException appears in logcat.

I can't claim to know Android development, but I traced the code back for `isBroadcomChip`:
https://github.com/seemoo-lab/nexmon/blob/456923f409212603ba5631c9eb346496216ca8c9/app/app/src/main/java/de/tu_darmstadt/seemoo/nexmon/utils/FirmwareUtil.java#L25-L28

https://github.com/seemoo-lab/nexmon/blob/456923f409212603ba5631c9eb346496216ca8c9/app/app/src/main/java/de/tu_darmstadt/seemoo/nexmon/utils/Nexutil.java#L189-L190

https://github.com/seemoo-lab/nexmon/blob/456923f409212603ba5631c9eb346496216ca8c9/app/app/src/main/java/de/tu_darmstadt/seemoo/nexmon/utils/Nexutil.java#L85-L87

https://github.com/seemoo-lab/nexmon/blob/456923f409212603ba5631c9eb346496216ca8c9/app/app/src/main/java/de/tu_darmstadt/seemoo/nexmon/utils/Assets.java#L49-L87

It seems like it isn't able to find `nexutil` in the apps assets/ directory.

I also get 'Permission denied' errors in logcat when attempting **Menu** > **Tools** > tick **nexutil** > **Install Binaries and Libraries**. May be worth checking that too.

I pulled the APK and unzipped it and couldn't find any binaries/executables in the assets folder. After finding an old APK at [seemoo-lab/bcm-public/tree/master/androidstudioprojects/Nexmon/app](https://github.com/seemoo-lab/bcm-public/tree/master/androidstudioprojects/Nexmon/app) I unzipped that and did find a `assets/nexmon` folder containing `nexutil` and the other binaries/libraries/firmware.

I don't know if Nexmon downloads these files or similar in the newer version of the app, so can't say  this is the actual issue. How is this just an issue now, seeing the [APK from the Nethunter Store](https://store.nethunter.com/en/packages/de.tu_darmstadt.seemoo.nexmon/) was released 2019-07-01?

This may all be completely off the mark, I don't know enough.

Jeklah, 2020-05-26:
I am also having this issue on Nexus 7 2013  Android 6.0.1.
What's the fix? Is there one?

ctrl-freak, 2020-06-02:
Previous issues indicate that it's probably best to go all out and build from source:

https://github.com/seemoo-lab/nexmon/issues/195

I grabbed a copy of the 2.1.3 APK from the web and unzipped it. I put the assets/nexmon contents into the `context.getFilesDir().getPath();` directory (in my case `/data/data/de.tu_darmstadt.seemo.nexmon/files`) and the app is still crashing when attempting to 'Grant Root, Permission & Check Installation' with the same error. I'm considering repackaging the APK, resigning and zipaligning but that effort would probably be best spent setting up a build environment.

Installing the older version from downloaded APK didn't have these issues, but I wasn't able to get it to install the correct firmware for my device.

Considering this is distributed in the Nethunter Store and the changelog on the app page says "Built for the NetHunter App Store" I'm guessing this is actually a build issue on the NetHunter side of things and building from source straight from this repository would get better support.

I've also got a Alfa AWUS036NEH on the way, so will not likely pursue this any further.

e2002e, 2020-10-12:
@ctrl-freak
You said: "Previous issues indicate that it's probably best to go all out and build from source:"
Not true, doesn't work either with the same exact issue.

jeancaffou, 2020-10-12:
Trying an old APK (2.1.3) kind of works, but it makes compatibility issues with other tools because of different versions of FW and tools.

Jeklah, 2020-10-12:
> @ctrl-freak
> You said: "Previous issues indicate that it's probably best to go all out and build from source:"
> Not true, doesn't work either with the same exact issue.

Thanks for testing what I didn't have time to verify.

> Trying an old APK (2.1.3) kind of works, but it makes compatibility issues with other tools because of different versions of FW and tools.

Kind of works is better than not at all! I will try this if I ever get some spare time again.

e2002e, 2020-12-31:
This is still unsolved ? 
I tried to run my branch 'notrack' but dependencies are no longer availables...
Maybe this would be fixed by not upgrading gradle ?
Anyway official master needs to be investigated upon.