seemoo-lab/nexmon #400: How to set frame injection bitrates on Nexus5?

holyens, 2020-05-05:
Hi, 

Thank you very much for your works! I have realized frame injection using nexmon on Nexus5, but I encountered a problem when I tried to set bitrates using the `iw` command. Once I execute the `iw dev wlan0 set bitrates xxx` command, then nexus5 will automatically reboot. I guess I should use the `nexutil` command to do it, but I did not find the option to set bitrates in the help of the `nexutil` command. Can you give me some help on setting up bitrates?

Best regards.
holyens

matthiasseemoo, 2020-05-05:
Bitrates for outgoing frames can be defined on a per frame basis using the radiotap headers:
https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/injection.c#L131 <https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/injection.c#L131>

289536718, 2020-05-08:
@holyens 
Hello, brother
How do you implement frame injection? Can you tell me the steps? Thank you.

matthiasseemoo, 2020-05-08:
https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/ioctl_4xx.c#L137 <https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/ioctl_4xx.c#L137>
https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/injection.c#L74 <https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/injection.c#L74>
https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/sendframe.c#L56 <https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/sendframe.c#L56>

Good luck. You can also inject from the host.

289536718, 2020-05-08:
@matthiasseemoo 
Thank you, sir. I have seen your reply in other questions, but I have never used anything like this before. How to use these functions? Do you want to cooperate with third-party software?

holyens, 2020-05-08:
> @holyens
> Hello, brother
> How do you implement frame injection? Can you tell me the steps? Thank you.

I achieve frame injection by using raw socket. This is my code [beacon.c](https://paste.ubuntu.com/p/Q8jZX77svW/)  which is from [issue#137](https://github.com/seemoo-lab/nexmon/pull/137#issuecomment-338541920).
It can be run by using `LD_PRELOAD=libnexmon.so`.  According to my understanding, libnexmon.so overrides the `write` function to enable frame injection (by calling the `inject_frame` in ioctl_4xx.c). The following is the corresponding code:
<https://github.com/seemoo-lab/nexmon/blob/master/utilities/libnexmon/nexmon.c#L355>
<https://github.com/seemoo-lab/nexmon/blob/d5fd58656942d770e59764d88130b09d0a701c77/patches/include/nexioctls.h#L47>
<https://github.com/seemoo-lab/nexmon/blob/d5fd58656942d770e59764d88130b09d0a701c77/utilities/libnexio/libnexio.c#L122>
<https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/injection.c#L74>

Using inject_frame function to send packets would be a more direct approach, but l don't know how to set `wlc` argument for inject_frame function now.

289536718, 2020-05-08:
@holyens 
thank you very much. I'll try as you and Mr. matthiasseemoo suggest. If there is a problem, I will ask for your help.

matthiasseemoo, 2020-05-08:
Just as a clarification, libnexmon is especially needed on devices that are connected over pice, as here frames are directly copied between the D11 core and the host and not the whole frame shows up in the ARM processor. Hence, we are using a hook to functions that write to a socket to instead send the injected frame using an ioctl. The bcm4339 is connected over SDIO, hence, here you can also inject frames through the regular path that is also used to send Ethernet frames from the host to the ARM firmware.

matthiasseemoo, 2020-05-08:
In our jammer demo project, you can also find some code to generate frames with different data rates and bandwidths:
https://github.com/seemoo-lab/wisec2017_nexmon_jammer_demo_firmware/blob/master/src/ioctl_5xx.c#L489 <https://github.com/seemoo-lab/wisec2017_nexmon_jammer_demo_firmware/blob/master/src/ioctl_5xx.c#L489>

289536718, 2020-05-09:
@matthiasseemoo 
Thank you very much, sir. But now I have a problem:

After loading nexmon or nexmon_csi, my RI4B doesn't seem to find any 5g signals. Whether through `iw wlan0 scan` or `airdump ng wlan0mon`, there is no 5g signal display.After I collected the CSI of 0x80, I still couldn't collect the CSI of 0x88, even if I used the ping command to add load to the router.

What's the matter? Will it affect my use of frame injection, because I want to collect CSI of data frame of 80MHz.

tjunet, 2020-05-14:
> Just as a clarification, libnexmon is especially needed on devices that are connected over pice, as here frames are directly copied between the D11 core and the host and not the whole frame shows up [...]

Thanks for your correction, your answer solved many of my confusions. Now I have a new problem. I found some printf statements in [ioctl_4xx.c](https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/ioctl_4xx.c#L75) and [ Injection.c](https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/injection.c#L178). but I can't get the output through `adb shell` on Nexus 5 because they seem to be running on BCM4339 firmware. Now I want to print some debugging message in `itctl_xxx.c`. Is there a way for me to obtain this output information?

matthiasseemoo, 2020-05-14:
> … <x-msg://20/#>

dhdutil consoledump

YaohuaLee, 2020-05-29:
@holyens
Hello, can you give me some header packages for you to compile beacon.c, I am missing too much when compiling here.
Thank you very much.

holyens, 2020-05-29:
> @holyens
> Hello, can you give me some header packages for you to compile `beacon.c`, I am missing too much when compiling here.
> Thank you very much.

No additional header files are needed to compile the `beacon.c` file. What you need to do is to  write `Android.mk` and `Makefile` for beacon.c, and then use `make` to build the executable file. In the nexmon/utilities directory, you can find some examples of Android.mk and Makefile. In addition, I have updated beacon.c ([click here](https://paste.ubuntu.com/p/CJXkJpZwnj/)) to support nexmon radiotap, which may be helpful to you.

YaohuaLee, 2020-05-29:
@holyens 
Thank you very much for your reply, but I have never been exposed to this aspect before. If I want to learn how to write `makefiles` and `Android.mk`  .This may be too much time, but my tutor doesn't give me much time.Although this is very presumptuous, if you can provide me with `makfile`, `Android.mk` ,I would be very grateful.

holyens, 2020-05-29:
@YaohuaLee 
You can download [beacon.zip](https://drive.google.com/open?id=1Y0_NGDVl8sMZNSraSEQgD599l0bUaUO3) and unzip it to the nexmon/utilities directory,  then  run:
```bash
cd nexmon/utilities/beacon
make 
make install
```
you may need to export the `NDK_ROOT` environment variable and execute `source setup_env.sh` before running the above commands.

YaohuaLee, 2020-05-30:
@holyens 
Thank you very much, sir. 
I have loaded the `beacon` into my nexus 5 now, but how to use it?   Running directly prompts or `LD_PRELOAD=libfakeioctl.so beacon`，it shows `delay = 0 number = 1 period = 0 ratespec = 0xc1000000 size = 80.`  I want to use nexus5 to implement frame injection for my ac-86u. Please forgive my stupidity and I hope you can give me an answer.

289536718, 2020-06-08:
@holyens 
Sir, how should I use the code you provided? I can run it directly in the adb shell, but I don't know the channel that generates the signal and can't receive it at the receiver.
I'd like to make precise settings. Could you give me some suggestions?

holyens, 2020-06-08:
> @holyens
> Sir, how should I use the code you provided? I can run it directly in the adb shell, but I don't know the channel that generates the signal and can't receive it at the receiver.
> I'd like to make precise settings. Could you give me some suggestions?

You need to use the `-k` parameter of `nexutil` to set the channel and bandwidth. There are two ways to specify the channel and bandwidth: `uint16` and `string`. you can find the relevant details in the comments of the [nexmon/utilities/nexutil/bcmwifi_channels.c](https://github.com/seemoo-lab/nexmon/blob/master/utilities/nexutil/bcmwifi_channels.c) file.

If you cannot see the signal, you can try to use a different channel.  In my attempts, I found that injection works well in the 2.4G band (such as 11 channels), but in the 5GHz band (such as 157 channels), it often does not work. I haven't found the reason so far. If anyone can help me, I would be very grateful.

289536718, 2020-06-08:
@holyens 
Thank you, sir. I'll try as you say.

DennisHerell, 2021-02-25:
Hi @holyens, sorry to ask in an old post, thanks for sharing your code for frame injection. Using the code you share, I manage to send out beacon frame and check it using tcpdump. However, I can't seem to extract the csi from these frame, is it possible to extract it? Also, it's possible to send other frame type right? just need to made the frame

Monjur-Mahathir, 2021-09-24:
Hi @holyens @DennisHerell I am working with Nexus 6p and Asus router to inject frames and collect CSI data. I can see the frame using tcpdump in the Nexus 6p device. But this frame is not found by the ASUS router. Can you kindly share with me what may be the problem? I just see the injected frame on the device that is injecting the frame, not in any other device. If you could help me out that would be greatly appreciated.