seemoo-lab/nexmon #402: bcmdhd native monitor patch

SamuelNigga, 2020-05-07:
Hi dear @matthiasseemoo .
My device xperia z5 compact and it uses bcm43455 .
i wanna try native monitor by using this patch: https://github.com/ruleh/misc/blob/master/monitor/bcmdhd_enable_monitor.patch

here is my device's kernel source:
https://github.com/LineageOS/android_kernel_sony_msm8994

so if i use bcmdhd monitor patch for my device (in kernel source dir 'patch bcmdhd_enable_monitor.patch') and compile kernel then flash image, the native monitor mode can work ? 
regards..

einsteinov, 2020-05-08:
It's working on Nexus 6p.But injection not working.

SamuelNigga, 2020-05-09:
> It's working on Nexus 6p.But injection not working.

what you mean by saying "its working" ?

einsteinov, 2020-05-10:
native monitor  with ruleh patch.you must have kernel 3.10.73

SamuelNigga, 2020-05-11:
> native monitor with ruleh patch.you must have kernel 3.10.73

i have 3.10.84, so if you press:

aireplay-ng -9 wlan0

what it say ?

closed 2020-05-16 by SamuelNigga