seemoo-lab/nexmon #403: Monitor mode doesn't work on RPi 4 Raspbian

ThatWeirdAndrew, 2020-05-07:
I have been unable to get monitor mode up and running with nexmon on my Raspberry Pi 4. 

I followed the steps of building the 7.45.154 patch for the Raspberry Pi 4 on the readme. After creating the monitor mode interface and running `ifconfig mon0 up`, I tried to test it with the command `airodump-ng mon0` however no access points showed up. I also tried running `nexutil -m1` but then `nexutil -m` returned 0 which indicated that it did not work. 

I then tried to get it to load the modified driver on boot by following the steps in the readme. After booting, trying to create the monitor interface resulted in the error: `command failed: Operation not supported (-95)`. 

I have also tried this running the kali images (both armhf and arm64) and I run into the same problems. I don't know what information is needed to troubleshoot further so any help and suggestions are appreciated.

ThatWeirdAndrew, 2020-05-21:
The problem seems to have been resolved. Monitor mode now works on Kali 2020.2.

closed 2020-05-21 by ThatWeirdAndrew

yq11, 2020-10-16:
How did you solve it, I also encountered the same problem