seemoo-lab/nexmon #404: with 0x88, no packet is captured

mikeWangJC, 2020-05-08:
Hey guys,  I meet some problem in using this interesting tool, hope someone can help me to find the answer.

1. With the 0x80,  the receiver can captures packets, but receiver can never capture packet wih 0x88. So, why is this? 

2. The collected CSI data usually covers only 20MHz. By viewing other issues, I understand the reason for this phenomenon. However, is there any way to make sure that I can collect CSI with bandwidth of 80MHz stably?

Compared to the Intel 5300 CSI tool and Atheros (I have used both tools), the advantage of this nexmon tool is that it can collect CSI with a larger bandwidth. When I use the Intel 5300 CSI tool, I can control the frequency and bandwidth of the signal, so I can collect the CSI I want, I mean with specific frequency and bandwidth. 

However, in nexmon, I failed to collect stable CSI with bandwidth of 80MHz on 5G.

So, I  hope someone can help me. Thanks in advance! :)

mzakharo, 2020-05-08:
I have had success capturing ICMP ping packets @ 80Mhz. I found sometimes you have to play with the payload size (increase it) to get the packets to appear. I found It is AP dependent how 0x88 packets behave.  Alternatively, you can  go with the packet injection route, and control bandwidth/channel manually.

289536718, 2020-05-08:
@mzakharo 
I also encountered the same problem, so what should I do? Specific steps?

mikeWangJC, 2020-05-08:
@mzakharo 
Hi buddy, thank you very much for your reply!!!!
The problem is how to realize the packet injection route???
Do hope you can help me with this! 
THANK YOU SO MUCH!

mzakharo, 2020-05-08:
Never needed to play with frame injection -> ping works reliably with my AC-68U AP and was simple enough to get what I needed.  From what I understand, any NIC that supports frame injection will do -> you can use scapy to craft the message. If you want to use broadcomm, it seems [4339 firmware patch](https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/injection.c ) has extra bells and whistles-> not only you can control data rate, but set periodicity and number of packets to inject, which will be guaranteed to be precise by the firmware.

289536718, 2020-05-08:
4339 firmware patch .Is mobisys2018_nexmon_software_defined_radio?@mzakharo   😂

mzakharo, 2020-05-08:
No, you probably dont need SDR. Try this: https://www.aircrack-ng.org/doku.php?id=aireplay-ng

289536718, 2020-05-08:
@mzakharo Thank you, sir. I'll try what you say. I hope I can keep in touch here😂

mikeWangJC, 2020-05-08:
@mzakharo  
dude, itried
 ‘ping+IP of another AP on 5G’ 
on my  ASUS ac86u, but no packet is received. Can you help me with this?

mikeWangJC, 2020-05-09:
@mzakharo
hi, dude, as you said,how to increase the payload size?

289536718, 2020-05-10:
@mzakharo 
Sir, I want to use RI4 to inject frame into ac86u to collect 80MHz CSI. Use aireplay for injection. The command is: `aireplay-ng-9-e essid-a bssid-i eth5 mon0.` The ac-86u has installed nexmon_csi and enabled monitoring mode (eth5).

An error occurred:` IOCTL (siocginfindex) failed: no such device`
How can I solve it? I have switched two network cards to the same channel.😂

nabeelni, 2022-10-03:
I can detect ping around 2600 bytes ping -s 2600 <ip router>