seemoo-lab/nexmon #405: BCM4356 - Force multicast rate

GaraQuor, 2020-05-13:
Hello,

Do you think it is possible to force tx multicast rate (to ac rate) in ibss mode by modifying the firmware ?
I know my question is only partially related to nexmon but I don't know where to ask.

Thanks

matthiasseemoo, 2020-05-13:
You can set any rate you want with nexmon.

GaraQuor, 2020-05-13:
Are you talking about actually changing the value of the send rate in the firmware or talking about injection mode via nexmon ?

Because the bcm4356 is marked as not compatible with injection on the readme (Or it is possible but not yet implemented ?)

matthiasseemoo, 2020-05-13:
Injection is possible on every chip, however, not yet implemented. However, it might be easier to just set another rate within the firmware when you are normally communicating with the Wi-Fi chip.

GaraQuor, 2020-05-13:
Ok thank you, that's exactly what I was hoping for !
I will try to do that :)

GaraQuor, 2020-06-30:
Hello
I dumped the rom and decompiled it with Ghidra (I don't know why but with IDAPro it fails) and the funcs addresses in wrapper.c are correct. Now I want force tx rate for all the packets do you know wich func I need to hook for that ? Can I override the rate in wlc_sendctl ?

Nedardo, 2021-03-04:
Hey GaraQuor, did you ever figure this out? I'm looking to try to implement a similar feature.