seemoo-lab/nexmon #408: "Nexus 5" 1.Can't start monitoring mode: 2.How to use frame injection

YaohuaLee, 2020-05-26:
I have linstall nexmon and many tools from `utilities` into my Nexus5 .There are no problems during the install process. Use `LD_PRELOAD=libfakeioctl.so tcpdump -i wlan0 `is also no problem.
But I had the following problems:

1: After start root, I use the command `nexuti -Iwlan0 - m2` can't start monitor all the time, but I can successfully switch channels using `nexutil -k.`

2: I want to use my Nexus5 inject frame to Asus ac-86u（it‘s has been install the nexmon_csi). I have noticed some functions mentioned in the discussion area, such as `sendframe.c, injection.c ioctl.c  etc`.  But how can I use them? I mean to give some **specific steps**. I just contacted with this knowledge, and may not use them.

I hope someone can help me. Thank you very much！（^_^）

289536718, 2020-05-26:
i did not use frame injection.   My asus ac86u collects csi of my phone hotspot after specifying mac-address. When other phone visit me, new data frame will be generated

YaohuaLee, 2020-05-26:
@289536718  
Thank you for your reply. I've tried to` ping` my `TP-link` and successfully captured CSI (although the capture rate is very slow), but I have some specific requirements to implement frame injection, I can't do it. Even now, I can't start monitor in Nexus 5.

YaohuaLee, 2020-05-26:
Thank you for your reply. I've tried to `ping` my TP-link and successfully captured CSI (although the capture rate is very slow), but I have some specific requirements to implement frame injection, I can't do it. Even now, I can't start monitor in nexus 5.

289536718, 2020-05-26:
i haven't tried to start frame injection on nexus 5. Although i have been advised,i seldom touch Linux,so i can't be implemented.sorry

mikeWangJC, 2020-05-28:
Got the same problem,
hope someone can help.

closed 2020-06-08 by YaohuaLee