seemoo-lab/nexmon #420: Is it compatable with Nexus 6P on Android 10 with LoS?

root3d1, 2020-07-10:
Hi,
I have a nexus 6P which i installed nethunter on over the android 10 with LoS setup. Will the drivers work with this version. Im not that technical with things like this and saw on the list the nexus 6p and saw android 8 comparability but not android 10!

Any clarity would be great thanks!

closed 2020-07-15 by root3d1

TheMMcOfficial, 2020-10-14:
Check this:

https://gitlab.com/kalilinux/nethunter/build-scripts/kali-nethunter-devices/-/issues/236

yesimxev, 2021-08-22:
Also https://forum.xda-developers.com/t/rom-official-kali-nethunter-for-the-huawei-nexus-6p-los17-1.4079087

Supported, but EAPOL packet is malformed

TQMatvey, 2022-03-01:
> Also https://forum.xda-developers.com/t/rom-official-kali-nethunter-for-the-huawei-nexus-6p-los17-1.4079087
> 
> Supported, but EAPOL packet is malformed

mmm i tried on other android versions and i get malformed packets everywhere xD

yesimxev, 2022-03-02:
Need to look into the driver or patches. The libs seem fine, I'm not an expert at it so it will take time. I also have limited time too