seemoo-lab/nexmon #423: RPi Kernel Update

ecdye, 2020-07-21:
The latest update to the RPi Kernel switches to 5.4.y and that is not currently supported. Could support be added so that installation works properly on the latest kernel update?

matthiasseemoo, 2020-07-21:
To answer your question, it should be techically possible to support the 5.4.y kernel. The limiting factor is always time.

ecdye, 2020-07-21:
Well support is going to be broken on RPi's until support for the new kernel is added as 5.4.y is now being pushed to RPi's by default.

JLP04, 2020-07-24:
I have the same issue.

matthiasseemoo, 2020-07-24:
just take a look to the changes we made to the driver and adopt them for the driver in the new kernel.

ecdye, 2020-07-25:
I simply don't have the time or experience to do that. I was simply trying to make this known to your team as it appeared as I was implementing a function in a project I am working on.

jlinktu, 2020-08-28:
Added support for 5.4 https://github.com/seemoo-lab/nexmon/commit/83eeaadd97efb06c0885826efb772654c015918d

closed 2020-08-28 by jlinktu