seemoo-lab/nexmon #425: Problem with sending frames in 5 GHz channels (RBPi 4)

salomoneli, 2020-08-12:
Hi!
I'm using the [sendframe()](https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm43455c0/7_45_189/nexmon/src/sendframe.c) function to send raw WiFi frames from a Raspberry Pi 4. The CSI of those frames is then captured at another Raspberry Pi. This already works fine for all channels in the 2.4 GHz band. But when I try to send a frame in a channel of the 5 GHz band, it doesn't work anymore. 

For sending the frames I use the [send_packet()](https://github.com/seemoo-lab/mobisys2018_nexmon_covert_channel/blob/de4e3823eb2262ae1391e2aa80d513e6700adeed/src/ioctl_7xx.c#L344) function, but instead of sendframe_with_timer() I simply called sendframe().
And for setting the transmission rate I used something similar to what is suggested [here](https://github.com/seemoo-lab/wisec2017_nexmon_jammer_demo_firmware/blob/master/src/ioctl_5xx.c#L489).

I already tried different transmission rates and packets, but nothing changed.
I've read through the issues regarding frame injection and it seems that also others have a problem with sending frames in the 5GHz channels.

Does anyone have any ideas what the problem might be and help me with this issue?

Thanks in advance!

salomoneli, 2020-08-17:
**Update**: 
Sending frames and extracting the CSI works also for these channels now (format: _channel/bandwidth_):
**36**/20, **40**/20, **44**/20 and **48**/20

I changed the setting for the transmission rate _from this:_

```
    data_rate = RATES_RATE_6M;
    switch (modulation)
    {
        case 2: // 802.11n
        {
            rate = RATES_OVERRIDE_MODE | RATES_ENCODE_HT | RATES_HT_MCS(data_rate) | (ldpc ? RATES_LDPC_CODING : 0);
        }
        break;

        case 3: // 802.11ac
        {
            rate = RATES_OVERRIDE_MODE | RATES_ENCODE_VHT | RATES_VHT_MCS(data_rate) | RATES_VHT_NSS(1);
        }
        break;
    }

    switch (bandwidth)
    {
        case WL_CHANSPEC_BW_20:
        {
            rate |= RATES_BW_20MHZ;
        }
        break;

        case WL_CHANSPEC_BW_40:
        {
            rate |= RATES_BW_40MHZ;
        }
        break;

        case WL_CHANSPEC_BW_80:
        {
            rate |= RATES_BW_80MHZ;
        }
        break;
    }

    return rate;
 ```

_To this:_

```
    rate = RATES_RATE_6M;
    switch (bandwidth)
    {
        case WL_CHANSPEC_BW_20:
        {
            rate |= RATES_BW_20MHZ;
        }
        break;

        case WL_CHANSPEC_BW_40:
        {
            rate |= RATES_BW_40MHZ;
        }
        break;

        case WL_CHANSPEC_BW_80:
        {
            rate |= RATES_BW_80MHZ;
        }
        break;
    }

    return rate;
```

But for all other 5 GHz channels it is still not working.

matthiasseemoo, 2020-08-18:
you need to add all the channels you want to use to:
https://github.com/seemoo-lab/nexmon_csi/blob/master/src/regulations.c#L195 <https://github.com/seemoo-lab/nexmon_csi/blob/master/src/regulations.c#L195>

salomoneli, 2020-08-18:
Thanks for your response!
I added them to the file, and now the channels 149, 153, 157, 161, 165 are working too.
```
#if (NEXMON_CHIP == CHIP_VER_BCM43455c0)
    0xe02a, // 36/80
    0xe29b, // 157/80
    CH20MHZ_CHSPEC(52),
    CH20MHZ_CHSPEC(56),
    CH20MHZ_CHSPEC(60),
    CH20MHZ_CHSPEC(64),
    CH20MHZ_CHSPEC(100),
    CH20MHZ_CHSPEC(104),
    CH20MHZ_CHSPEC(108),
    CH20MHZ_CHSPEC(112),
    CH20MHZ_CHSPEC(116),
    CH20MHZ_CHSPEC(120),
    CH20MHZ_CHSPEC(124),
    CH20MHZ_CHSPEC(128),
    CH20MHZ_CHSPEC(132),
    CH20MHZ_CHSPEC(136),
    CH20MHZ_CHSPEC(140),
    CH20MHZ_CHSPEC(149),
    CH20MHZ_CHSPEC(153),
    CH20MHZ_CHSPEC(157),
    CH20MHZ_CHSPEC(161),
    CH20MHZ_CHSPEC(165),
#endif
```

However, the problem remains for the channels 52 to 140.
Setting them with nexutils -k works fine, i.e. no error appears when setting the chanspec (neither on the transmitter nor on the receiver side).

For example:
Set `nexutil -k48/20` on both sides
Initiate to send frame 
Receiver gets frame and extracts CSI
=> works as it should

Set `nexutil -k52/20` on both sides
Initiate to send frame
=> nothing happens

The only thing that changes is the chanspec, everything else stays the same.

matthiasseemoo, 2020-08-18:
Not all channels can be set by the driver, so it is normally better to define an ioctl to set a channel directly in the firmware. especially, if you want to have channels with larger bandwidth.

salomoneli, 2020-08-19:
I've already written an ioctl where I set the chanspec using the [set_chanspec()](https://github.com/seemoo-lab/nexmon/blob/e90e3937d3f25fcb521afa938ef4176ff627c6f0/patches/common/helper.c#L265) function, but it makes no difference. 
Or is there another way to set the channel in the firmware?
(By the way: I only need 20MHz bandwidth for all channels in the 2.4 GHz and the 5 GHz band)

tomer-avrahami, 2020-09-05:
Hi @Salomoneli,

Did you manage to extract CSI and send packets simultaneously using RBPi4?
Can you please share your setup and the needed modifications?

Thanks,
Tomer

salomoneli, 2020-09-15:
@tomer-avrahami: yes, I can send packets and extract the CSI from them. but you should know that the payload of the packets gets destroyed during the CSI extraction. so you can't use the payload to transmit information if you want to extract the CSI.

-) Activate the CSI extraction on a device as described [here](https://github.com/seemoo-lab/nexmon_csi#usage).
-) For sending frames I looked at the code segments suggested [here](https://github.com/seemoo-lab/nexmon_csi/issues/88#issuecomment-642501530)
-) I am using an ioctl to trigger the transmission of a packet from one RBPi (server) to another (client)
-) At the client the CSI is extracted from this packet
-) The client responds with an acknowledgement packet from which the server can extract the CSI
-) To trigger the transmission of the acknowledgement packet (after the packet from the server has been received) I modified the [process_frame_hook()](https://github.com/seemoo-lab/nexmon_csi/blob/ba99ce12a6a42d7e4ec75e6f8ace8f610ed2eb60/src/csi_extractor.c#L169) function.

I am sorry that I cannot go into much more detail with my description. I'm quite busy at the moment and still trying to find out why the packets are not sent when one of the DFS channels (52 to 140) is used. However, I hope it can help you anyway!

@matthiasseemoo: do you have any further suggestions why sending packets in the DFS channels isn't working?

tomer-avrahami, 2020-09-19:
@salomoneli Thanks a lot for your answer and detailed links!
I am currently working on the implementation.
I will try packet injection of DFS channel once I have a stable setup.

Colin-Guolin, 2020-12-11:
> Hi!
> I'm using the [sendframe()](https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm43455c0/7_45_189/nexmon/src/sendframe.c) function to send raw WiFi frames from a Raspberry Pi 4. The CSI [...]

Hi, my brother

You and I are doing the same thing, I am trying to send frame from one raspberry pi and capture the CSI at another raspberry pi, I am struggling to figure out how to send frame using Nexmon for many weeks, since I am new to Linux system. Could you please tell the step that how do you implement sending frames using nexmon if you have time for this, it would be greatly helpful for me. Thanks in advance

Best regards,
Colin

salomoneli, 2020-12-16:
The issue was that the aforementioned channels in the 5 GHz band are so-called DFS channels and there are restrictions if devices want to initiate a transmission on these channels.

I found a solution for this and described it here: https://github.com/seemoo-lab/nexmon/issues/443#issuecomment-746715730

closed 2020-12-16 by salomoneli