seemoo-lab/nexmon #426: Setting 5GHz channels on Nexus6P

Deltaaaaaaaaaa, 2020-08-15:
Thankyou for this amazing tool! I'm working on frame injection on 160MHz bandwidth using Nexus6P.
When setting the 5GHz channels (using  command "nexutil -k" )，it worked well on channel149, 153, 157, 161, 165. But when setting other 5GHz channels, an error occurs.
The supported 5GHz channels does not support 160MHz bandwidth. To obtain 160MHz CSI measure, BCM4358 needs to be set to channels such as 64, 128 .etc which can't be set correctly.

1. Injection test using command "aireplay-ng -9" works normally on 2.4GHz channels and channel 149, 153, 157, 161, 165.

2. Here's the error: 
```
angler:/ # ifconfig wlan0 up                          
angler:/ # nexutil -k149/80                           
angler:/ # nexutil -k128/80                           
__nex_driver_io: error ret=-1 errno=22
angler:/ # nexutil -k128/160                          
__nex_driver_io: error ret=-1 errno=22
```

3. Information returned by command "nexutil -V": 
```
platform     |                    firmware                                | vendorid | deviceid |
------------ | --------------------------------------------- | ---------- | --------- |
Nexus 6P    | 7.112.300.14 (r707445) FWID 01-3242a45b |   0x14e4  |   0x43e9 |

radiorev  | chipnum | chiprev | chippackage | corerev | boardid | boardvendor |
---------- | ---------- | ------- | -------------- | ------- | --------- | -------------- |
0x2e2069 |  0x4358   |     0x3  |         0x2       |    0x30  |   0x7a1  |      0x14e4     |

boardrev | driverrev   | ucoderev   | bus | phytype | phyrev | anarev | nvramrev
  -------- | ----------- | ------------ | --- | --------- | ------- | -------- | --------
      P100 | 0x77012c0 | 0x3c3013d | 0x0 |     0xb    |   0x11 |    0x0    |  0x7a1f2
```

4. What might be the problem? What can I do to fix it? By the way, the current firmware version is 7.112.300.14, is it the same thing with 7.112.300.14.sta? Could this be the problem?

Really appreciate it if someone help me fix this!  T-T

matthiasseemoo, 2020-08-15:
Hi JingBo,

the bcm4358 does not support 160 MHz channels.
channels that are not working out of the box, first need to be activated in the regulations.c file:
https://github.com/seemoo-lab/nexmon_csi/blob/master/src/regulations.c <https://github.com/seemoo-lab/nexmon_csi/blob/master/src/regulations.c>
Currently, we did not yet release 160 MHz CSI extraction code.

Matthias

Deltaaaaaaaaaa, 2020-08-16:
Hi Mr. Matthias, 

Thankyou for your instruction! 
Does bcm4366c0 on AC-86U support 160MHz channels?
Is the 160 MHz CSI extraction released on other devices such as AC-86U?
Is it possible to obtain 160MHz CSI measurement base on the current release?

JingBo
[…](#)
> Hi JingBo, the bcm4358 does not support 160 MHz channels. channels that are not working out of the box, first need to be activated in the regulations.c file: [...]