seemoo-lab/nexmon #428: Wifi detecting but not connecting on Kali Linux on 8GB(raspberry pi)

rayavarapuvikram1, 2020-08-25:
When I am connecting to wifi it is asking password again and again. I reinstalled [Kali](https://www.offensive-security.com/kali-linux-arm-images/)  more than 10 times but it didn't work. Please, someone, help me.

e2002e, 2021-01-12:
Are you talking about connecting to a wifi hostspot for which you have the password ? Do you mean that installing nexmon broke your wifi ? Then try restoring the firmware backup... Otherwise I can't help you and you need to read more about wifi hacking.

rayavarapuvikram1, 2021-01-12:
My raspberry pi 4 is not connecting to my Wi-Fi router. My router is working fine.

e2002e, 2021-01-12:
Ok, so you did install the firmware as explained in the README ? 
This must be the wrong one, you need to restore it. No hardware is broken, only software.
On the last resort, reinstall the whole system.

rayavarapuvikram1, 2021-01-12:
I reinstalled whole system 4 times.
But issue isn't resolved

e2002e, 2021-01-12:
Did you make any change to the wifi firmware when you reinstalled the system ? This is kali linux ? most likely it doesn't support your wifi chip.
Could you connect even once to the hotspot beforehand ?
Can you first make the password appear when you type it to be really sure this is not some stupid mistake.
Also another test is reconnecting another device you have at home (forget the hospot and retype password) to make sure the wifi password has not been changed.

rayavarapuvikram1, 2021-01-12:
My password is password so it's not spelling mistake. 
No I am unable to connect at least once
No I didn't make any change

e2002e, 2021-01-12:
Do you have any other device for testing ? Maybe a friend or your neighbor can try to connect to see if it really comes from your device.
You said you could not connect even a single time and this is triggering.

rayavarapuvikram1, 2021-01-12:
I will record a video of problem and post here in sometime

e2002e, 2021-01-12:
This will not help, you must try with another device. You have a phone right ?

rayavarapuvikram1, 2021-01-12:
Yup

e2002e, 2021-01-12:
Please go into your wifi setting and choose 'forget wifi' then reconnect. At least we'll know it's not your router.

rayavarapuvikram1, 2021-01-12:
Connecting normally

e2002e, 2021-01-12:
OK, I suggest that you install a more stable system than kali.
Most of the tools that you find in it are available for ubuntu.
sqlmap / metasploit / aircrack ... and so on. And having to set these by yourself is an excellent way of knowing more and no getting stuck when things simply don't work.
Let us know whether wifi is fixed by changing the system.

rayavarapuvikram1, 2021-01-12:
It is working fine with Rasapian OS, Ubuntu, only problem is with kali

e2002e, 2021-01-12:
Oh ok, why are you posting this here ?
This is nexmon's github issues's page ... The problem is with kali so post on their forum if you really need to install it.

rayavarapuvikram1, 2021-01-12:
I didn't get a solution there also.
I am looking for help either from Nexmon or Kali

e2002e, 2021-01-12:
Kali is a bad system; if you want to have fun hacking just install sqlmap and metasploit, and a vpn, you don't need everything else.
Hacking wifi is impossible. For the hashes that you will get in the databases you can crack them with john the ripper.
Do yourself a favor and install these software on a clean ubuntu distro, which has much better stability.