seemoo-lab/nexmon #429: Raspbian kernel 5.X

dev-sora, 2020-08-31:
Hi,

Is there a plan to make Nexmon tool compatible with Raspbian kernel 5.X?

tomer-avrahami, 2020-09-04:
Worked for me over PI4, raspbian with kernel 5.4.

dev-sora, 2020-09-05:
Didn't you see any warning like "kernel version not supported" ?

tomer-avrahami, 2020-09-05:
Not that I notice.
I was able collect CSI data.

jdevelop, 2020-10-05:
@tomer-avrahami can you please share how did you build the patches for kernel 5.4 ?

There's no sources for 5.4 in https://github.com/seemoo-lab/nexmon/tree/master/patches/bcm43430a1/7_45_41_46/nexmon

axiomiety, 2020-11-01:
@jdevelop same issue here - 5.4 is listed under the bcm43455c0's Makefile but that's specific to the Pi 3B+. It doesn't seem to be supported for the Pi 3B, which is using the 7.45.98.94 firmware.

Does anyone know what that would involve? Happy to give it a shot.

tomer-avrahami, 2020-11-04:
@jdevelop Sorry I missed your message.
I am working with RPI4. As @axiomiety already mentioned, 5.4 sources exists for its driver:
https://github.com/seemoo-lab/nexmon/tree/master/patches/bcm43455c0/7_45_189/nexmon/brcmfmac_5.4.y-nexmon

dimage, 2020-11-16:
5.9?