seemoo-lab/nexmon #430: Extract CSI and send packets simultaneously

tomer-avrahami, 2020-09-04:
Hi,

First, thanks a lot for your work.
I want to build a setup of two devices which are able to communicate one with each other and to extract CSI data from those packets. Time between packets should be ~50ms. 
Do you think this setup possible using Raspberry PI hardware (any version)?
Do you have any recommendation how I can build this kind of setup?

I am currently working over Pi4B, Kernel 5.4.
I was not able to connect to a WiFi network while extracting CSI.

From what I understand from Nexmon documentation, it is not possible to inject packets over PI4\Pi3B+, while CSI extraction is not possible anymore using PI3B (which enables packets injection). Am I correct? 

Thanks,
Tomer

tomer-avrahami, 2020-11-07:
For future readers - I was able both to inject and extract CSI from PI4 using kernel 5.4.

closed 2020-11-07 by tomer-avrahami

Hodit01, 2021-01-28:
If you don't mind, could you tell me how?

Arsering, 2021-07-25:
@tomer-avrahami Sir, Would you mind tell me how to inject and extract CSI from PI4 simultaneously?

Arsering, 2021-07-25:
> For future readers - I was able both to inject and extract CSI from PI4 using kernel 5.4.
 Sir, Would you mind tell me how to inject and extract CSI from PI4 simultaneously?