seemoo-lab/nexmon #431: where to buy  wifi card bcm43455c0 for Raspberry Pi B3+/B4

baldevbarrsiwal, 2020-09-06:
Dear All, 

I am struggling a bit to get the required hardware **" wifi card bcm43455c0 for Raspberry Pi B3+/B4.** It would be a great help if you can provide any lead where I can buy this hardware. I have explored on many sites to buy it but could got any luck.

I'm sorry to bother you for such a basic question.
Best Regards
Baldev

zeroby0, 2020-09-11:
Hey, the chip is built in to Raspberry Pi 3B+/4B. 
You don't have to buy a wifi card, but you will need an Ethernet cable if you want to SSH into Pi over network.

baldevbarrsiwal, 2020-09-13:
Hi Arvind,

Thanks a lot for the information, I am asking if I would like to add additional wifi card bcm43455c0 for Raspberry Pi B3+/B4. That is a requirement for my project. any lead would be really appreciated. Thank you

matthiasseemoo, 2020-09-13:
Just attach more Raspberry Pis :-)

baldevbarrsiwal, 2020-09-30:
Hi,

I have got Pi 3B+,, when I run the following cmd, it gives me wifi chipset details as 43430 not the bcm 43455.  am I pointing to something wrong ? could someone please help?

`root@raspberrypi:/home/pi# sudo airmon-ng
PHY	Interface	Driver		Chipset

phy0	wlan0		brcmfmac	Broadcom 43430
`

clanec15, 2021-01-09:
I dont know if you still have this problem or not but the chipset is marked as a 43430 because thats the firmware that is being used

baldevbarrsiwal, 2021-01-09:
Hi Thanks a lot for the input, yeah it's bit misleading because the chip set is 43455. However, we can close the issue

closed 2021-01-09 by baldevbarrsiwal