seemoo-lab/nexmon #432: errors while running "make"

TheEcstacyKiller, 2020-09-16:
root@raspberrypi:/home/pi/nexmon# make
  BUILDING BUILDTOOLS
make[1]: Entering directory '/home/pi/nexmon/buildtools'
  BUILDING flash_patch_extractor
make[2]: Entering directory '/home/pi/nexmon/buildtools/flash_patch_extractor'
make[2]: 'fpext' is up to date.
make[2]: Leaving directory '/home/pi/nexmon/buildtools/flash_patch_extractor'
  BUILDING ucode_extractor
make[2]: Entering directory '/home/pi/nexmon/buildtools/ucode_extractor'
make[2]: 'ucodeext' is up to date.
make[2]: Leaving directory '/home/pi/nexmon/buildtools/ucode_extractor'
  BUILDING b43/assembler/b43-asm.bin
make[2]: Entering directory '/home/pi/nexmon/buildtools/b43/assembler'
     CC       b43-asm.bin
/usr/bin/ld: cannot find -ll
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:75: b43-asm.bin] Error 1
make[2]: Leaving directory '/home/pi/nexmon/buildtools/b43/assembler'
make[1]: *** [Makefile:21: b43/assembler/b43-asm.bin] Error 2
make[1]: Leaving directory '/home/pi/nexmon/buildtools'
make: *** [Makefile:9: buildtools] Error 2

Not exactly sure what's wrong here?

billard091, 2021-01-08:
Hello everyone,

i have the same issue.
Is there a solution yet?

thank you in advice!

zedrdave, 2021-01-21:
Not sure why, but `buildtools/b43/assembler/Makefile` was modified to use `-ll` instead of `-lfl`. Changing lines 27-28 to re-enable `-lfl` instead of `-ll` seems to fix the problem.

(you might also need to `apt install libfl-dev` for it to work)