seemoo-lab/nexmon #433: airodump-ng not working with Pi 4 and 7_45_206

meliodasren, 2020-09-20:
I got a Pi 4 and set it up with a fresh Raspbian OS. I tried following the instructions to install bcm43455c0 | 7_45_206. As far as I can tell it is installed fine. Yet when I try to run airodump-ng no APs are found.

I am hoping I just missed something but I do not know how to debig this further.

Any help/ideas are appreciated. Sorry for the formatting as I am on my phone.

Note: The packaged airodump-ng did the same thing. I found some post online suggesting that building it from source may help but it did not help. The version below was from building it from source.

root@raspberrypi:/home/pi# dmesg | grep nexmon
[    5.543882] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BC
M4345/6 wl0: Sep 19 2020 23:04:54 version 7.45.206 (nexmon.o
rg: 2.2.2-326-g050d-dirty-9) FWID 01-88ee44ea

root@raspberrypi:/home/pi# uname -a
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40
BST 2020 armv7l GNU/Linux

root@raspberrypi:/home/pi# airodump-ng --help

  Airodump-ng 1.6 rev 9fdc52f2 - (C) 2006-2020 Thomas d'Otreppe

root@raspberrypi:/home/pi# iw list | grep monitor
                 * monitor
                 * monitor: 0x00 0x10 0x20 0x30 0x40 0x50 0x
60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * monitor: 0x00 0x10 0x20 0x30 0x40 0x50 0x
60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * #{ managed } <= 1, #{ monitor } <= 1, #{
P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1,
                 * #{ managed } <= 1, #{ AP } <= 1, #{ monit
or } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,

root@raspberrypi:/home/pi# ifconfig mon0
mon0: flags=867<UP,BROADCAST,NOTRAILERS,RUNNING,PROMISC,ALLM
ULTI>  mtu 1500
        unspec DC-A6-32-B7-34-01-00-00-00-00-00-00-00-00-00-
00  txqueuelen 1000  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collis
ions 0

meliodasren, 2020-09-20:
I am hoping that I'm doing something wrong, but it looks like if I install 7_45_189 and then install 7_45_206 then it works for some reason.

I'm currently writing a script that I can run when my Pi 4 boots that will basically:
- install 7_45_189 (using "make install-firmware" on it)
- install 7_45_206 (using "make install-firmware" on it)
- put wlan0 into monitor mode

If I then run ifconfig I see Rx packets on mon0 and if I run airodump-ng I see something relatively quickly.

doragasu, 2020-11-17:
I think for Raspbian you should be using 1_45_189. According to documentation, 7_45_206 is for Raspberry Pi OS.