seemoo-lab/nexmon #434: Unable to change to channels on 80MHz bandwidth on Pi 4

Gi-z, 2020-09-22:
I have a Pi 4 running the latest version of Raspberry Pi OS (August 2020) on which I've installed nexmon using the instructions for bcm43455c0 hardware. This appears to be running properly and I have a separate monitor interface which is working properly now, which I can confirm by running airodump-ng. 

However, I am unable to change the channel and bandwidth to anything on 80MHz using nexutil.
[image]

No 80MHz channel seems to work, but other 20 and 40MHz channels work with no issue. Is there something I'm doing wrong?

matthiasseemoo, 2020-09-22:
You need to add your channel to the regulations.c file:
https://github.com/seemoo-lab/nexmon_csi/blob/master/src/regulations.c <https://github.com/seemoo-lab/nexmon_csi/blob/master/src/regulations.c>

You can change FW_VER_7_45_189 in  https://github.com/seemoo-lab/nexmon_csi/blob/master/src/regulations.c#L236 <https://github.com/seemoo-lab/nexmon_csi/blob/master/src/regulations.c#L236> to FW_VER_ALL as it is a flashpatch, assuming that you have enough flashpatches left in the newer firmware version.

Gi-z, 2020-09-23:
I have installed nexmon (as opposed to nexmon_csi) so as to use the frame injection functionality which is listed on the README. As far as I can tell, there is no regulations.c file in the nexmon source?

matthiasseemoo, 2020-09-23:
just copy it.

Gi-z, 2020-09-24:
Added the regulations.c file from nexmon_csi, and changed the FW_VER_7_45_189 reference to FW_VER_ALL. Compilation now fails at the LINKING OBJECTS stage. Is this indicating that I do not have enough flashpatches left in this firmware version? 

log/linker.err: 
```
/home/pi/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-ld: Removing unused section '.comment' in file 'obj/udptunnel.o'
/home/pi/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-ld: Removing unused section '.ARM.attributes' in file 'obj/udptunnel.o'
/home/pi/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-ld: Removing unused section '.text.tinflate_write_objmemx' in file 'obj/ucode_compression_code.o'
/home/pi/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-ld: Removing unused section '.text.tinflate_read_objmemx' in file 'obj/ucode_compression_code.o'
/home/pi/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-ld: Removing unused section '.text.wlc_ucode_write_compressed_args' in file 'obj/ucode_compression_code.o'
/home/pi/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-ld: Removing unused section '.text.wlc_ucodex_write_compressed_args' in file 'obj/ucode_compression_code.o'
/home/pi/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-ld: error: no memory region specified for loadable section `.text.wlc_valid_chanspec_ext'
```

matthiasseemoo, 2020-09-24:
Try to put FW_VER_ALL also in the following place:
https://github.com/seemoo-lab/nexmon/blob/master/patches/common/wrapper.c#L1779

> On 24. Sep 2020, at 15:51, Glenn Forbes <notifications@github.com> wrote:
> 
> text.wlc_valid_chanspec_ext

Gi-z, 2020-09-24:
This resolved the issue during building, however once the new driver is loaded I get the following output from the system upon calling nexutil.

```
root@raspberrypi:/home/pi/nexmon/patches/bcm43455c0/7_45_206/nexmon# nexutil
root@raspberrypi:/home/pi/nexmon/patches/bcm43455c0/7_45_206/nexmon# nexutil -k
Segmentation fault
root@raspberrypi:/home/pi/nexmon/patches/bcm43455c0/7_45_206/nexmon# 
Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.858608] Internal error: Oops: 207 [#4] SMP ARM

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859352] Process nexutil (pid: 29087, stack limit = 0xadade323)

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859374] Stack: (0xd44efd88 to 0xd44f0000)

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859397] fd80:                   d44efdb4 bf4e3da8 d44efd88 d8b8cc00 bf4ea3e4 00000540

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859427] fda0: 00000000 d8b8cc10 d44efdf4 d44efdb8 bf4c5304 bf4bef40 00000106 0000001c

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859456] fdc0: 0000002c 0000002c d44efdf8 c3c87400 c3c86c00 d7b7b480 0000002c d44efdf8

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859484] fde0: 00000000 c1204f88 d44efe2c d44efdf8 c095b844 bf4c5120 7fffffff 3d2982e2

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859513] fe00: c0959c54 c1204f88 d7b7b480 0000002c c3c86c00 d44efeb4 00000000 00000000

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859542] fe20: d44efe94 d44efe30 c095bb98 c095b6b0 d44efe24 00080000 02d40000 00000000

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859570] fe40: 00010000 00000000 00000000 0000002c d880ab40 00000000 0000719f 00000000

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859598] fe60: 00000000 3d2982e2 c041d0b8 00000000 c1204f88 d7ddf700 00000000 00000000

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859627] fe80: d44ee000 00000121 d44eff8c d44efe98 c08dc3c0 c095b8e8 d44efebc c0959bf4

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859655] fea0: c3c86c00 fffffff7 00000000 017f7398 0000002c 00000000 00000000 00000005

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859683] fec0: 00000000 00000000 d44efeb4 00000000 d7ddf700 00000000 00000000 00000000

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859711] fee0: c08dc0dc c095a190 d44eff54 00000000 00000000 00000010 00000000 00000000

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859739] ff00: 00000011 d44eff70 c3c85800 00000000 00000000 00000001 00000014 00000000

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859768] ff20: 00000000 3d2982e2 c041d0b8 c1204f88 00000000 d7ddef80 be8f2f78 be8f2f78

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859796] ff40: bf000000 00000126 d44effa4 d44eff58 c08dc7b0 c08e12d0 00000008 d44eff70

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859825] ff60: d44eff94 3d2982e2 00000014 00015188 00010158 00094dc8 00000121 c02011c4

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859853] ff80: d44effa4 d44eff90 c08dc46c c08dc2ec 00000000 00000000 00000000 d44effa8

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859881] ffa0: c0201000 c08dc450 00015188 00010158 00000003 017f7398 0000002c 00000000

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859910] ffc0: 00015188 00010158 00094dc8 00000121 00000000 00000000 00000000 be8f2f64

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.859938] ffe0: 00000000 be8f2f08 000142d0 0002ef04 60000010 00000003 00000000 00000000

Message from syslogd@raspberrypi at Sep 24 22:28:36 ...
 kernel:[93741.860813] Code: e24dd00c e52de004 e8bd4000 e1a04000 (e5900000) 
```

evertismael, 2021-05-05:
Any news on this error?, I'm encountering the same issue.
@Gi-z  Did you manage to make it work for any channel of 80Mhz?
I'd really appreciate the help