seemoo-lab/nexmon #437: Pi 4 automatic loading driver after reboot

davidbenucci, 2020-10-15:
For Pi4, the instructions for loading modified driver after reboot are the same of Pi3? In this case, how can I connect to an access point in parallel without wpa_supplicant? Thanks in advance.

baldevbarrsiwal, 2020-11-12:
You can copy manually the installed patch to  ***brcmfmac*** folder after patching `make backup-firmware` 

check your kernel version and then 

`cp /home/pi/nexmon/patches/bcm43455c0/7_45_206/nexmon/brcmfmac_5.4.y-nexmon/brcmfmac.ko /home/pi`
`cp /home/pi/brcmfmac.ko /lib/modules/5.4.51-v7l+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/`

or 

`cp /home/pi/ brcmfmac.ko /lib/modules/5.4.51-v7+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/`

michalisk13, 2023-01-22:
Hi,

I have similar issue.
After building the patch succesfully, there is no brcmfmac.ko file created which I can copy to the `/lib/modules/5.15.84-v8+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/`

In my patch folder I have the following files created after the build process: 

```
 root@raspberrypi:/home/pi/nexmon/patches/bcm43455c0/7_45_206/nexmon# ls
brcmfmac43455-sdio.bin  brcmfmac43455-sdio.bin.orig  BUILD_NUMBER  gen  log  Makefile  obj  patch.ld  src
```

Please note that the patch looks to be working.
The problem is to add persistancy when reboot. 

Thanks,
Mak

sknbl, 2023-02-11:
what i did to make the rpi4 load the modified driver after reboot is:

# mv /lib/modules/<kernel release>/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko.xz /lib/modules/<kernel release>/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko.xz.orig

# cp /home/pi/nexmon/patches/driver/brcmfmac_<branch>-nexmon/brcmfmac.ko /lib/modules/<kernel release>/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/

# xz brcmfmac.ko

# reboot