seemoo-lab/nexmon #451: I need some help with implementing frame injection on Nexus 5.

zmsenqn, 2020-12-20:
Hi, I am kind of new to this wireless field. So please forgive if I made some mistake.

I've been struggling for long time to implement frame injection.

My configuration is as follows:

I use a Nexus 5 as a transmitter, and use a Raspberry Pi B3+ as a receiver.
(So I install Nexmon firmware on the Nexus 5, and install Nexmon_csi firmware on the Raspberry Pi)

Based on my understanding, I need to inject frames on Nexus 5 and my Raspberry Pi should collect CSI - generate pcap file for later analysis.

After I install the firmware on my Nexus 5 (i.e., execute 'make install-firmware' in '/nexmon/patches/bcm4339/6_37_34_43/nexmon' directory), I configure the wireless channel and bandwidth, using 'nexutil -k[CH]/[BW]'. When I check the configuration with 'nexutil -k', the configuration looks good.

Now the problem begins.

I searched and referred as many tips and hints I could see on this issue thread, but still my Raspberry Pi fails to acquire any frames from the Nexus 5.

Primarily, I referred C codes suggested in 
https://github.com/seemoo-lab/nexmon/pull/137#issuecomment-338541920
and 
https://github.com/seemoo-lab/nexmon/issues/400#issuecomment-625946764 , 
however, whenever I execute the compiled C program on the Nexus 5, zero packets are captured on Raspberry Pi's 5500 port.
(On the Raspberry Pi, I did as follows:
1. Install nexmon_csi firmware
2. Add mon0 interface
3. Execute 'makecsiparams' by providing my Nexus 5's MAC address as a parameter
4. Execute 'nexutil' by providing the returned string in Step 3 as a parameter
5. Run 'tcpdump -i wlan0 dst port 5500 -vv')

I also checked the output of 'tcpdump -i wlan0' on Raspberry Pi, and what I could see is that bunch of beacon packets inbound from the surrounding APs (I could see SSIDs from the legitimate APs in the captured packet, but none of them were related to my Nexus 5).

Whenever I execute the C program on my Nexus 5, I applied LD_PRELOAD as follows:
LD_PRELOAD=libfakeioctl.so [MY_INJECTION_PROGRAM] -[PARAMETERS] ,
however, still I could not observe any changes on my Raspberry Pi (I also tried this command after executing 'setenforce 0' with su permission).

I suspect that the raw socket-based approach in my current C code is not working properly - although I am using exactly same logic in https://github.com/seemoo-lab/nexmon/issues/400#issuecomment-625946764.

Please help me to implement the frame injection program. I will be more than happy if anyone can provide me a piece of code that can inject a single frame on Nexus 5.

Thank you for your help, and reading this long post.

matthiasseemoo, 2020-12-26:
> #137 (comment) <https://github.com/seemoo-lab/nexmon/pull/137#issuecomment-338541920>
> #400 (comment) <https://github.com/seemoo-lab/nexmon/issues/400#issuecomment-625946764> ,
> I suspect that the raw socket-based approach in my current C code is not working properly - although I am using exactly same logic in #400 (comment) <https://github.com/seemoo-lab/nexmon/issues/400#issuecomment-625946764>.

Maybe check first with another receiver in monitor mode that you successfully injected frames.

libnexmon.so and libfakeioctl.so allow you to inject frames from user space of your operating system for example by using airplay-ng.

matthiasseemoo, 2020-12-26:
by the way, we also have an app that allows to easily inject frames using a nexus 5, it is part of our reactive jamming demo:
https://github.com/seemoo-lab/wisec2017_nexmon_jammer_demo_app <https://github.com/seemoo-lab/wisec2017_nexmon_jammer_demo_app>

Feel free to check it out. In the Transmitter mode, you can inject UDP packets with a configured modulation. In the Receiver mode you can check with a secondary Nexus 5 whether your frames were received correctly.

The demo paper can be found here:
https://www.seemoo.tu-darmstadt.de/mschulz/wisec2017_massive_reactive_jamming_demo.pdf <https://www.seemoo.tu-darmstadt.de/mschulz/wisec2017_massive_reactive_jamming_demo.pdf>

You can also find our poster here (might take some time to load as it is the print version):
https://drive.google.com/file/d/0By-JLU-GbHBjcFdld0xFMEJDdDQ/view?usp=sharing <https://drive.google.com/file/d/0By-JLU-GbHBjcFdld0xFMEJDdDQ/view?usp=sharing>

Matthias

zmsenqn, 2021-01-01:
@matthiasseemoo Thank you so much for your tip. I finally made it work. Now I can collect CSI with the Jammer app, and the firmware generated by Nexmon CSI.

closed 2021-01-01 by zmsenqn

ywy809080, 2022-04-12:
@zmsenqn 
Hello.
You have implemented the frame injection function on nexus 5. I don't know how to write the sending code. Can you give me some guidance?

ywy809080, 2022-04-18:
@matthiasseemoo 
I'm done with the "Getting Started" part on the Nexus 5. Then I need to implement frame injection on the Nexus 5. How do I write the code?
The execution is complete on the mobile phone
adb shell
su
What do I need to do next?
Thank you for your help!

matthiasseemoo, 2022-04-18:
on the nexus 5 you can just inject frames with any tool that can do this. just set LD_PRELOAD=libnexmon.so.
If you want to inject frames from within the firmware, you can also take a look at our nexmon jammer app code.
I already referenced it a couple of times in other issues.

jlinktu, 2024-11-05:
iirc the jammer uses [this](https://github.com/seemoo-lab/wisec2017_nexmon_jammer_demo_firmware/blob/8eb32a8b96bf42b6e2c4a0ffa4db77a8cf04ebc8/src/ioctl_5xx.c#L178) wlan frame as template, that would mean that the frame's destination address is 'NEXMON' (`4e:45:58:4d:4f:4e`) and it's source address is 'JAMMER' (`4a:41:4d:4d:45:52`). Filtering on the CSI extractor for the Nexus 5's wlan interface address won't give you the results you are looking for. You wan't to filter for the template's source address instead.

Stefanzcj, 2024-11-05:
> iirc the jammer uses [this](https://github.com/seemoo-lab/wisec2017_nexmon_jammer_demo_firmware/blob/8eb32a8b96bf42b6e2c4a0ffa4db77a8cf04ebc8/src/ioctl_5xx.c#L178) wlan frame as template, that would [...]

Thank you very much for your reply,
As what you have said, the destination of the frames sent from the phone has been specified, in my understanding, I should also make sure that the MAC adress of the receiver(Raspberry pi) is 'NEXMON' (4e:45:58:4d:4f:4e) . Do I understand it correctly? If it's true, how can I make sure that the MAC adress is 'NEXMON'. 

Thanks again

jlinktu, 2024-11-05:
No, the RPi is in monitor mode, it will process any frame seen on the medium, no matter the destination address.

Stefanzcj, 2024-11-06:
> No, the RPi is in monitor mode, it will process any frame seen on the medium, no matter the destination address.

Dear jlinktu

I have just tried to change the source mac adress 'Jammer', but the receiver still got nothing. I am wondering if there is some other things wrong with my instructions. Can you help me? 

Best.

jlinktu, 2024-11-06:
Instead of using a CSI extractor, use a device in monitor mode first to check for the frames sent out by the Jammer App to make sure they are what you expect.