seemoo-lab/nexmon #458: How to get rom.bin from Pi 4B?

Hodit01, 2021-01-25:
# 1.Background：
**I want extract rom.bin from Pi 4B，I got some information in the README, but I don’t know how to implement it in details.**
[image]
**And I also saw**
[image]
**in https://github.com/seemoo-lab/bcm-rpi3/tree/master/firmware_patching/rom_extraciton**
**I think maybe I can modify some parameters in the code and use it on Pi 4B，but I don't know how to use it.**
# 2.Help
## 1. How to get rom.bin in Pi 4B?  hope it can be more detailed...
## 2. Where can I find more information about it?

**If anyone can provide some help, I would really appreciate that.**

jlinktu, 2021-01-25:
You can use this [rom_extraction](https://github.com/seemoo-lab/nexmon/tree/master/patches/bcm43455c0/7_45_154/rom_extraction) patch for the bcm43455c0.
If you want to use it with kernel version 5.4, just copy [brcmfmac_5.4.y-nexmon](https://github.com/seemoo-lab/nexmon/tree/master/patches/bcm43455c0/7_45_206/nexmon/brcmfmac_5.4.y-nexmon) to the patch directory and replace `4.9` with `5.4` in the `Makefile`.

Then follow the instructions to [install](https://github.com/seemoo-lab/nexmon#build-patches-for-bcm43430a1-on-the-rpi3zero-w-or-bcm434355c0-on-the-rpi3rpi4-using-raspbianraspberry-pi-os-recommended) and [use](https://github.com/seemoo-lab/nexmon#dumping-a-clean-rom-after-copying-to-ram) the patch.

Hodit01, 2021-01-27:
> You can use this [rom_extraction](https://github.com/seemoo-lab/nexmon/tree/master/patches/bcm43455c0/7_45_154/rom_extraction) patch for the bcm43455c0.
> If you want to use it with kernel version [...]

thanks,bro! I get it

closed 2021-01-27 by Hodit01