seemoo-lab/nexmon #459: csi data

carolinagarciaguerrero, 2021-01-27:
someone get data like that?
[image]

I think my data is wrong, can someone help me?

jlinktu, 2021-01-27:
The data seems correct.
You are tuned to a 80MHz wide channel and thus extract CSI over 256 subcarriers. However, it seems you only see transmissions on 20MHz control channels that are subbands of the 80MHz channel, leaving you with valid CSI over 64 of the 256 subcarriers.
Please have a look at the [FAQ](https://nextcloud.seemoo.tu-darmstadt.de/apps/files_sharing/accept/ocinternal:489):
> If half or three quarters of extracted CSI for 40 or 80MHz are mostly empty or contain very low values it is very likely that you received a 20MHz frame inside a 40 or 80MHz wide channel. As 40 and 80MHz wide channels are just several bonded 20MHz channels this is totally possible and valid. One of the bonded 20MHz channels will be used as control channel to transmit control frames. Hence, you probably captured CSI of control frames or the transmitter is simply not using the available bandwidth.

sarielkk, 2023-08-30:
> 有人得到这样的数据吗？ [image]
> 
> 我认为我的数据有误，有人可以帮助我吗？

Hello! I have the same problem with you. Did you solve the problem?