seemoo-lab/nexmon #460: Could not set 5GHz Channels >  Channel 140  on RPI4 Model B

wildpeartree, 2021-01-31:
Hi,

When I try to set my RPI4 wlan0 interface to channel 149 (iw wlan0 set channel 149), I faced following message "brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=57499, -52 " (it happens when _desired channels_ > 140)

My country reg is US, therefore all supported channel is enable. OS: Kali Linux 2020.4

What can I do for fix this problem, thanks.

Best

jlinktu, 2021-02-01:
Removing restrictions and/or adding channels to firmware and/or driver might help you, have a look at https://github.com/seemoo-lab/nexmon/issues/443#issuecomment-748666226

wildpeartree, 2021-02-15:
> Removing restrictions and/or adding channels to firmware and/or driver might help you, have a look at [#443 (comment)](https://github.com/seemoo-lab/nexmon/issues/443#issuecomment-748666226)

Hi,

This work for removing restrictions and adding channels issue thanks a lot. 

But after that, another problem comes up. After this update,  I can't capture any packets. When I analyse syslog, I face this warning message periodically  "brcmfmac: brcmf_escan_timeout: timer expired" 

I used to have default nexmon driver image for RPI4 , Kali Linux 2020.4 , Kernel: 4.19.118-v7l ( my first version, there is no problem about monitor mode and capturing data, but couldn't set 80 MHz channels and couldn't set channel > 140 )

Following step by step driver install instructions, compile and install new driver version: 7_45_189 ( removing restrictions and adding channels ok, but can't capture any packets.)

Is there any idea for that ?

Best,