seemoo-lab/nexmon #461: Truncated packets for bcm4358

johnpaul7, 2021-01-31:
I have been testing nexmon with the Nexus 6P. I have got it working with several different firmware versions and multiple versions of Android, successfully managing to capture packets in monitor mode and to perform packet injection, but for some reason in every case it always seems to truncate the 4-way WPA handshake packets.

For each of the 4 packets I only get the first 86 bytes (or 114 if I have radiotap headers on) with a highlighted 'Malformed packet' error in Wireshark. When capturing the same packets on a separate device I get packet sizes of 133, 155, 189 and 133 respectively. The packets agree for the first 86 bytes but for some reason I'm not getting the rest of the packets.

There is a similar bug mentioned in issue #231 for the Samsung Galaxy S7. Do you think this is the same behaviour? In that case @matthiasseemoo suggested it might be splitting the packets and sending them directly to the host. I searched the firmware for the string splitrx but didn't find any occurrences. If this is happening, theoretically it should be possible to capture the rest of the packet at the android kernel level yes? I'd be happy to look into this if I can be reassured I'm headed in the right direction?

I have tried debugging the monitor_hook function with dhdutil consoledump but it seems the rest of the packets are really not in the firmware. I have not noticed that any other kind of packets are malformed or truncated (although I haven't done any thorough testing). Could this perhaps be some kind of WPA security measure?

Out of curiosity, I have seen several people making claims that they have gotten monitor mode working on the Nexus 6P, but can anyone confirm that they have actually captured entire EAPOL frames? And if so which firmware/android versions they were using. I noticed that airodump-ng has the little message in the top right corner that it captured a WPA handhake which is misleading as if you try to open the file afterwards it says 0 handshakes captured.

nutrikazuma, 2021-02-02:
Hi there @johnpaul7 . Look, I'm starting with this a few days ago, so I don't know how to help with your problem, but maybe you can help me with mine. I see you have a lot of experience with the nexmon over bcm4358 (just the same on my phone, Nexus 6p) so I like to ask you if you can send me the nexmon apk you are using, because I try two versions, 2.2.2-226-ga988-dirty from nethunter repository that force close itself in start, and 2.1.3 i get from internet that starts but don't work because ask me about an updated 2.2.2-348-g4ccc to work with my actual firmware, but this update I don't find anywhere. Can you give me a help, please? Thanks in advance.

johnpaul7, 2021-02-04:
@nutrikazuma I am not using the nexmon apk. After flashing the patched firmware I am enabling monitor mode with the `nexutil` binary and then running `tcpdump` and `airodump-ng` on the command line prefixed with the `LD_PRELOAD=libnexmon.so` command.

yesimxev, 2022-02-10:
I'm glad you opened this issue. We also have the problem, using kalilibnexmon on NetHunter with LOS18.1. I'll paste some more info when I'm in front of PC