seemoo-lab/nexmon #466: Building nexmon app error

Oniat, 2021-02-26:
Hi, im trying to build the app from nexmon/app directory but getting this problem on libpcap,

```make[6]: Leaving directory '/home/tisn/Downloads/nexmon/app/app/src/main/external/src/libmyaircrackwep'
make -C libwireshark_helper/.
make[6]: Entering directory '/home/tisn/Downloads/nexmon/app/app/src/main/external/src/libwireshark_helper'
cd ../libpcap && make
make[7]: Entering directory '/home/tisn/Downloads/nexmon/app/app/src/main/external/src/libpcap'
/home/tisn/Android/SDK/NDK/android-ndk-r11c/ndk-build NDK_APPLICATION_MK=`pwd`/Application.mk NDK_APP_OUT=. TARGET_PLATFORM=android-21
make[8]: Entering directory '/home/tisn/Downloads/nexmon/app/app/src/main/external/src/libpcap'
make[8]: *** No rule to make target '../../../../../../../utilities/libpcap/pcap-can-linux.c', needed by 'local/armeabi/objs/pcap/./__/__/__/__/__/__/__/utilities/libpcap/pcap-can-linux.o'. Stop.
make[8]: Leaving directory '/home/tisn/Downloads/nexmon/app/app/src/main/external/src/libpcap'
make[7]: *** [Makefile:5: libs/armeabi/libpcap.a] Error 2
make[7]: Leaving directory '/home/tisn/Downloads/nexmon/app/app/src/main/external/src/libpcap'
make[6]: *** [Makefile:9: ../libpcap/local/armeabi/libpcap.a] Error 2
make[6]: Leaving directory '/home/tisn/Downloads/nexmon/app/app/src/main/external/src/libwireshark_helper'
make[5]: *** [Makefile:7: libwireshark_helper/.] Error 2
make[5]: Leaving directory '/home/tisn/Downloads/nexmon/app/app/src/main/external/src'
make[4]: *** [Makefile:2: all] Error 2
make[4]: Leaving directory '/home/tisn/Downloads/nexmon/app/app/src/main/external'
make[3]: *** [Makefile:90: libs/armeabi] Error 2
make[3]: Leaving directory '/home/tisn/Downloads/nexmon/app/app/src/main'
make[2]: *** [Makefile:2: all] Error 2
make[2]: Leaving directory '/home/tisn/Downloads/nexmon/app/app/src'
make[1]: *** [Makefile:2: all] Error 2
make[1]: Leaving directory '/home/tisn/Downloads/nexmon/app/app'
make: *** [Makefile:2: all] Error 2
```

e2002e, 2022-01-16:
before building the utilities, you must run 'source setup_env.sh' and 'make' in the root nexmon folder.

Oniat, 2022-11-24:
This issue was caused by improper setup and and some requirements not being met.

closed 2022-11-24 by Oniat