seemoo-lab/nexmon #469: Add support for RPi Kernel 5.10.y

Nightkurz, 2021-03-22:
New default kernel for raspberry pi after issuing the apt-get update && apt-get upgrade is 5.10.y, is there a posibility to add support?

BlobbyBob, 2021-04-10:
Is there any way of building Nexmon at the moment on Raspberry Pi OS? 

While I'm able to download an older image which uses the 5.4 kernel, the raspberrypi-kernel-headers seem to be only available for the 5.10 kernel (at least from the official package sources).

sondree, 2021-04-14:
I just finished the compilation process and got stuck on this too :(

sondree, 2021-04-16:
> Edit: After some more hours invested, I found, that the packages are still available [on the server](https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/), but just not listed by apt. So you at least still build and install nexmon with these older packages.

For those of you that that face the same and just want to get this built and running. The following worked for me:

> wget https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20200902-1_armhf.deb
> wget https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20200902-1_armhf.deb 
>apt remove --purge raspberrypi-kernel-headers
>dpkg -i raspberrypi-kernel_1.20200902-1_armhf.deb
>dpkg -i raspberrypi-kernel-headers_1.20200902-1_armhf.deb

As always, on your on risk. I successfully compiled "nexmon/patches/bcm43455c0/7_45_206/nexmon" using the above packages.

Polymathical, 2021-06-11:
> For those of you that that face the same and just want to get this built and running. The following worked for me:
>
>
> As always, on your on risk. I successfully compiled [...]

And?

LuViKu, 2021-06-11:
For me it doesn't continue after the APPLYING PATCHES part. Do you encounter the same error?

mildsunrise, 2021-09-15:
@jlinktu said they're working on support for 5.10.

In the meanwhile I've submitted a minimal PR in #495.
I have tested it for a while and it seems to work without problems.

mildsunrise, 2021-09-16:
@jlinktu Also, with [two changes](https://github.com/mildsunrise/nexmon/compare/kernel-5.10...mildsunrise:dkms) to the codebase, the drivers can be deployed as DKMS modules, which dramatically improves the experience:

~~~ bash
cp -r .../brcmfmac_5.10.y-nexmon /usr/src/nexmon-brcmfmac-1.0
dkms install -a nexmon-brcmfmac/1.0
~~~

This builds & installs the driver, and does it for all kernels. It also loads it afterwards.
And most importantly, it will automatically be rebuilt after kernel upgrades.

Would you be interested in merging this?

sandro-byte32, 2021-09-29:
@mildsunrise is it possible to adopt these changes for the bcm43430a1 firmware. 
So the Raspberry Pi Zero can also be used with the 5.10 Kernel?

mildsunrise, 2021-09-29:
I haven't checked but I think the drivers are identical across firmwares... you can try