seemoo-lab/nexmon #477: Unable to build Nexmon APK (error: 'LINUX_USB_MON_DEV' undeclared)

muskfan, 2021-04-28:
Hi,

I have followed these links to solve some of my issues, but I'm still unable to create the Nexmon APK because of the listed down error log

* Links followed -
  1. **README.md** instructions at https://github.com/seemoo-lab/nexmon#build-patches-for-bcm4330-bcm4339-and-bcm4358-using-a-x86-computer-running-linux-eg-ubuntu-1604
  2. Final `make` command in *$NEXMON_ROOT/utilities* directory is completed successfully
  3. Went into *$NEXMON_ROOT/app* directory then executed `make` to generate Nexmon APK
  4. Tried these [instructions](https://www.gitmemory.com/issue/seemoo-lab/nexmon/398/621723388) but now stuck on the below error log

* Error received when building - 

```text
...

 cd ../libpcap && make
make[7]: Entering directory '/home/lubuntu/nexmon/app/app/src/main/external/src/libpcap'
/home/lubuntu/Android/Sdk/ndk/android-ndk-r11c/ndk-build NDK_APPLICATION_MK=`pwd`/Application.mk NDK_APP_OUT=. TARGET_PLATFORM=android-21
make[8]: Entering directory '/home/lubuntu/nexmon/app/app/src/main/external/src/libpcap'
[armeabi] Compile thumb  : pcap <= pcap-linux.c
[armeabi] Compile thumb  : pcap <= pcap-usb-linux.c
././../../../../../../../utilities/libpcap/pcap-usb-linux.c: In function 'usb_findalldevs':
././../../../../../../../utilities/libpcap/pcap-usb-linux.c:282:29: error: 'LINUX_USB_MON_DEV' undeclared (first use in this function)
   pcap_strlcpy(usb_mon_dir, LINUX_USB_MON_DEV, sizeof usb_mon_dir);
                             ^
././../../../../../../../utilities/libpcap/pcap-usb-linux.c:282:29: note: each undeclared identifier is reported only once for each function it appears in
././../../../../../../../utilities/libpcap/pcap-usb-linux.c: In function 'usb_activate':
././../../../../../../../utilities/libpcap/pcap-usb-linux.c:641:42: error: 'LINUX_USB_MON_DEV' undeclared (first use in this function)
   pcap_snprintf(full_path, USB_LINE_LEN, LINUX_USB_MON_DEV"%d", handlep->bus_index);
                                          ^
././../../../../../../../utilities/libpcap/pcap-usb-linux.c:641:59: error: expected ')' before string constant
   pcap_snprintf(full_path, USB_LINE_LEN, LINUX_USB_MON_DEV"%d", handlep->bus_index);
                                                           ^
/home/lubuntu/Android/Sdk/ndk/android-ndk-r11c/build/core/build-binary.mk:460: recipe for target 'local/armeabi/objs/pcap/./__/__/__/__/__/__/__/utilities/libpcap/pcap-usb-linux.o' failed
make[8]: *** [local/armeabi/objs/pcap/./__/__/__/__/__/__/__/utilities/libpcap/pcap-usb-linux.o] Error 1
make[8]: Leaving directory '/home/lubuntu/nexmon/app/app/src/main/external/src/libpcap'
Makefile:5: recipe for target 'libs/armeabi/libpcap.a' failed
make[7]: *** [libs/armeabi/libpcap.a] Error 2
make[7]: Leaving directory '/home/lubuntu/nexmon/app/app/src/main/external/src/libpcap'
Makefile:9: recipe for target '../libpcap/local/armeabi/libpcap.a' failed
make[6]: *** [../libpcap/local/armeabi/libpcap.a] Error 2

....
```

Thank you for going through this issue.

muskfan, 2021-04-29:
I have applied the following patch and was able to successfully execute `make` in *$NEXMON_ROOT/app* directory, then used `./gradlew build` to generate the APKs and using the *debug* apk for now in my phone. *How this patch impacts/degrades the original intended functionality, I don't know since I haven't tested the app yet.* **The developers can leave a comment about it.** This whole effort was done because of the [Nexmon](https://store.nethunter.com/en/packages/de.tu_darmstadt.seemoo.nexmon/) app on the Kali Nethunter store was crashing on Nexus 5.

```diff
diff --git a/app/app/src/main/external/src/libpcap/Android.mk b/app/app/src/main/external/src/libpcap/Android.mk
index 081055dc..bbe3047e 100644
--- a/app/app/src/main/external/src/libpcap/Android.mk
+++ b/app/app/src/main/external/src/libpcap/Android.mk
@@ -10,25 +10,22 @@ include $(CLEAR_VARS)
 
 LOCAL_SRC_FILES := \
   $(UTILITIES_PATH)/libpcap/pcap-linux.c \
-  $(UTILITIES_PATH)/libpcap/pcap-usb-linux.c \
-  $(UTILITIES_PATH)/libpcap/pcap-can-linux.c \
   $(UTILITIES_PATH)/libpcap/pcap-netfilter-linux.c \
   $(UTILITIES_PATH)/libpcap/pcap-netfilter-linux-android.c \
   $(UTILITIES_PATH)/libpcap/fad-gifc.c \
   $(UTILITIES_PATH)/libpcap/pcap.c \
-  $(UTILITIES_PATH)/libpcap/inet.c \
   $(UTILITIES_PATH)/libpcap/gencode.c \
   $(UTILITIES_PATH)/libpcap/optimize.c \
   $(UTILITIES_PATH)/libpcap/nametoaddr.c \
   $(UTILITIES_PATH)/libpcap/etherent.c \
+  $(UTILITIES_PATH)/libpcap/fmtutils.c \
   $(UTILITIES_PATH)/libpcap/savefile.c \
   $(UTILITIES_PATH)/libpcap/sf-pcap.c \
-  $(UTILITIES_PATH)/libpcap/sf-pcap-ng.c \
+  $(UTILITIES_PATH)/libpcap/sf-pcapng.c \
   $(UTILITIES_PATH)/libpcap/pcap-common.c \
-  $(UTILITIES_PATH)/libpcap/bpf/net/bpf_filter.c \
+  $(UTILITIES_PATH)/libpcap/bpf_filter.c \
   $(UTILITIES_PATH)/libpcap/bpf_image.c \
   $(UTILITIES_PATH)/libpcap/bpf_dump.c \
-  $(UTILITIES_PATH)/libpcap/version.c \
   mysniffer.c \
   myinjecter.c
```

Given the current repo uses older gradle, for the `./gradlew build` command to run successfully, install a older version of JDK -
```bash
sudo apt install openjdk-8-jdk
```

If you get the warning for missing NDK then create a *local.properties* in *$NEXMON_ROOT/app* with the below contents -
```properties
ndk.dir=/path/to/your/own/ndk/directory/android-ndk-r11c
```

#### [Untested nexmon debug build 20210429](https://github.com/muskfan/attachments/blob/main/nexmon-debug-build-20210429.apk) - *APK version: 2.2.2-340-g297e-dirty*

Similar/References - #397 #424 #435 #453

closed 2021-04-29 by muskfan

tranzmatt, 2022-04-29:
Was the LINUX_USB_MON_DEV resolved?  I'm getting this same error when trying to build the app.

muskfan, 2022-05-01:
> Was the LINUX_USB_MON_DEV resolved? I'm getting this same error on Ubuntu 20.04 when trying to build the app. I had installed all the same pcap packages.

Hi, I was able to resolve it only after changing build code and modifying make file, as mentioned in my above [patch comment](https://github.com/seemoo-lab/nexmon/issues/477#issuecomment-829212527). I could not built it, without making any changes. 

_P.S.: How this patch impacts/degrades the original intended functionality, I don't know since I haven't tested the app yet._