seemoo-lab/nexmon #480: Galaxy Note 3 Nexmon breaks WiFi

lavanoid, 2021-06-09:
Hello. I'm using a Galaxy Note 3 (BCM4339) and I installed the Nexmon firmware via Hijacker (https://github.com/chrisk44/Hijacker), onto Lineage OS (currently using 18.1/Android 11, and this issue also exists on LineageOS 17.1/Android 10).

Monitor mode and packet injection works fine, however it breaks standard Android WiFi functionality entirely. I can scan for networks but I cannot connect to them. I need to manually revert the firmware to the one that LineageOS comes with, each time I want to use the WiFi adapter for normal things.

I have some dmesg output, which I hope can help:
```
[ 1937.663668] CFG80211-TRACE) wl_escan_handler : SCAN COMPLETED: scanned AP count=8
[ 1937.725429] msm_thermal:msm_therm_temp_log Debug Temp for Sensor: 
[ 1937.725464] msm_thermal:msm_therm_temp_log (0 --- 47)(1 --- 46)(2 --- 46)(3 --- 47)(4 --- 47)(5 --- 49)(6 --- 47)(7 --- 47)(8 --- 47)(9 --- 46)(10 --- 45)dhd_wl_ioctl: WLC_IOCTL: cmd: 23, ret = -17
[ 1937.755252] dhd_wl_ioctl: WLC_GET_VAR: mfp, ret = -23
[ 1937.755259] CFG80211-INFO2) wl_set_set_cipher : Get MFP failed! Check MFP support in FW 
[ 1937.755267] CFG80211-INFO2) wl_cfg80211_connect : Invalid ciper
```

ROM source: https://download.lineageos.org/hlte

I hope there's a fix for this, and I am comfortable with modifying/recompiling my kernel if need be.
Thanks!

matthiasseemoo, 2021-06-09:
What is the version of your original Wi-Fi firmware? Just run strings on it and look at the last line. Maybe the driver requests more featuers than available in our old firmware version.

lavanoid, 2021-06-09:
Had to do some digging in the `/system` partition but I think I finally found the firmware file. 

```
/system/vendor/firmware # strings ./bcm4335_V0111.0559.hcd | head -20                                                                                                                                                                                                               
BCM4339 4.1+ H-LTE SemcoB67-0111
```

What confuses me is that the file name starts with `bcm4335` but the files header has `BCM4339`. Not sure if that matters though.

matthiasseemoo, 2021-06-09:
hcd is the bluetooth firmware file.

lavanoid, 2021-06-09:
In which case, I have no idea where the firmware is. This is all I can find in the firmware directory:
[n3_vendor_firmware.zip](https://github.com/seemoo-lab/nexmon/files/6627484/n3_vendor_firmware.zip)

```
[lava@LLL firmware]$ ls -1
a330_pfp.fw
a330_pm4.fw
bcm2079xB4_firmware.ncd
bcm2079xB4_pre_firmware.ncd
bcm2079xB5_firmware.ncd
bcm2079xB5_pre_firmware.ncd
bcm4335_V0111.0559.hcd
cmnlib.b00
cmnlib.b01
cmnlib.b02
cmnlib.b03
cmnlib.mdt
cpp_firmware_v1_1_1.fw
cpp_firmware_v1_1_6.fw
cpp_firmware_v1_2_0.fw
keymaster
n3_vendor_firmware.zip
widevine.b00
widevine.b01
widevine.b02
widevine.b03
widevine.mdt
[lava@LLL firmware]$ pwd
/home/lava/Documents/N3Stuff/system/vendor/firmware
[lava@LLL firmware]$ 
```

lavanoid, 2021-06-10:
I turned off the WiFi, re-enabled it and took another dmesg output. Some of this looks interesting.

```
[  366.447342] init: processing action (wlan.driver.status=ok) from (/vendor/etc/init/hw/init.qcom.rc:378)
[  366.447560] init: starting service 'macloader'...
[  366.449998] 
[  366.450000] Dongle Host Driver, version 1.141.64.59 (security_patch)
[  366.450003] Compiled in drivers/net/wireless/bcmdhd on Jun  9 2021 at 02:35:45
[  366.450008] wl_android_wifi_on in
[  366.450014] wifi_platform_set_power = 1
[  366.450018] ------------------------------------------------
[  366.450021] ------------------------------------------------
[  366.450026] brcm_wlan_power Enter: power on
[  366.450034] [brcm_wlan_power] Bluetooth Power On.
[  366.453696] msm_pcm_playback_prepare: perf: 1
[  366.453702] __q6asm_open_write: session[1] wr_format[0x0]
[  366.453707] __q6asm_open_write: session[1] open.mode_flags:10000000
[  366.460580] msm_pcm_playback_prepare: session ID 1
[  366.465236] hostwake line change [BT]
[  366.467830] bluesleep_sleep_work - uport is null [BT]
[  366.501810] WL_REG_ON on-step : [0]
[  366.501822] WL_REG_ON on-step-2 : [1]
[  366.513781] init: Service 'macloader' (pid 6552) exited with status 0 oneshot service took 0.065000 seconds in background
[  366.513832] init: Sending signal 9 to service 'macloader' (pid 6552) process group...
[  366.514003] libprocessgroup: Successfully killed process cgroup uid 1000 pid 6552 in 0ms
[  366.635150] [brcm_wlan_power] BT_REG_OFF.
[  366.838532] sdio_reset_comm():
[  366.865782] mmc0: queuing unknown CIS tuple 0x80 (2 bytes)
[  366.868739] mmc0: queuing unknown CIS tuple 0x80 (7 bytes)
[  366.870398] mmc0: queuing unknown CIS tuple 0x80 (3 bytes)
[  366.932702] msmsdcc_execute_tuning : set tuning_done ture.
[  366.937199] F1 signature OK, socitype:0x1 chip:0x4339 rev:0x1 pkg:0x0
[  366.937787] DHD: dongle ram size is set to 786432(orig 786432) at 0x180000
[  366.938482] dhdsdio_download_firmware: firmware path=/vendor/etc/wifi/bcmdhd_sta.bin, nvram path=/vendor/etc/wifi/nvram_net.txt
[  367.012443] dhdsdio_write_vars: Download, Upload and compare of NVRAM succeeded.
[  367.183697] dhd_bus_init: enable 0x06, ready 0x06 (waited 0us)
[  367.186798] dhd_wl_ioctl: WLC_GET_VAR: cisdump, ret = -23
[  367.186831] [WIFI_SEC] dhd_check_module_cid: CIS reading failed, ret=-23
[  367.198484] Firmware up: op_mode=0x0005, MAC=f0:a6:f5
[  367.198943] [WIFI_SEC] sec_get_param_wfa_cert: File [/data/.roamoff.info] doesn't exist 
[  367.201588] dhd_wl_ioctl: WLC_SET_VAR: const_awake_thresh, ret = -23
[  367.201597] dhd_preinit_ioctls set const_awake_thresh failed -23
[  367.205426] dhdcdc_set_ioctl: SET PM to 2
[  367.205740] [WIFI_SEC] sec_control_pm: /data/.psm.info open failed, so set PM to 2
[  367.206117] [WIFI_SEC] sec_get_param_wfa_cert: File [/data/.bustxglom.info] doesn't exist 
[  367.208163] [WIFI_SEC] sec_get_param_wfa_cert: File [/data/.frameburst.info] doesn't exist 
[  367.209463] dhd_wl_ioctl: WLC_SET_VAR: aibss_bcn_force_config, ret = -23
[  367.209473] dhd_preinit_ioctls Set aibss_bcn_force_config to 500, 5000, 5000 returned (-23)
[  367.210377] dhd_wl_ioctl: WLC_SET_VAR: pspretend_threshold, ret = -23
[  367.210384] dhd_preinit_ioctls pspretend_threshold for HostAPD returned (-23)
[  367.223322] Firmware version = wl0: Feb 26 2018 16:58:54 version 6.37.32.48.8 (A0 Station/P2P)
[  367.224177] type=1400 audit(1623283416.983:181): avc: denied { read } for comm="e.legacy" name=".wifiver.info" dev="mmcblk0p26" ino=16 scontext=u:r:hal_wifi_default:s0 tcontext=u:object_r:system_data_root_file:s0 tclass=file permissive=1
[  367.224484] [WIFI_SEC] sec_save_wlinfo: .wifiver.info File open failed.
[  367.225711] [WIFI_SEC] sec_get_param_wfa_cert: File [/data/.proptx.info] doesn't exist 
[  367.227471] type=1400 audit(1623283416.983:181): avc: denied { open } for comm="e.legacy" name=".wifiver.info" dev="mmcblk0p26" ino=16 scontext=u:r:hal_wifi_default:s0 tcontext=u:object_r:system_data_root_file:s0 tclass=file permissive=1
[  367.229034] dhd_wlfc_hostreorder_init(): successful bdcv2 tlv signaling, 64
[  367.229592] dhd_wl_ioctl: WLC_GET_VAR: pfnlbest, ret = -23
[  367.231301] dhd_wl_ioctl: WLC_GET_VAR: cisdump, ret = -23
[  367.231331] [WIFI_SEC] dhd_check_module_mac: Check module mac by legacy FW : f0:a6:f5
[  367.231538] [WIFI_SEC] Check Mac address in .mac.info 
[  367.232437] CFG80211-INFO2) wl_cfg80211_attach_post : p2p0: p2p_dev_addr=f2:a6:f5
[  367.309611] wl_create_event_handler(): thread:wl_event_handler:19ae started
[  367.309699] CFG80211-INFO2) wl_event_handler : tsk Enter, tsk = 0xea44c5fc
[  367.312936] audit: audit_lost=280 audit_rate_limit=5 audit_backlog_limit=64
[  367.312948] audit: rate limit exceeded
[  367.314811] type=1300 audit(1623283416.983:181): arch=40000028 syscall=54 per=800008 success=yes exit=0 a0=a a1=8914 a2=bef7ac58 a3=bef7ac44 items=8 ppid=1 ppcomm=init auid=4294967295 uid=1010 gid=1010 euid=1010 suid=1010 fsuid=1010 egid=1010 sgid=1010 fsgid=1010 tty=(none) ses=4294967295 comm="e.legacy" exe="/system/vendor/bin/hw/android.hardware.wifi@1.0-service.legacy" subj=u:r:hal_wifi_default:s0 key=(null)
[  367.317667] type=1302 audit(1623283416.983:181): item=0 name=(null) inode=4197 dev=b3:17 mode=0100644 ouid=0 ogid=0 rdev=00:00 obj=u:object_r:vendor_configs_file:s0
[... cut ...]
[  371.364783] init: Service 'vendor.nfc_hal_service' (pid 6618) exited with status 0
[  371.364821] init: Sending signal 9 to service 'vendor.nfc_hal_service' (pid 6618) process group...
[  371.364980] libprocessgroup: Successfully killed process cgroup uid 1027 pid 6618 in 0ms
[  371.611837] Governor: enter mode 0x2
[  371.712408] [SSP]: hub_batt_adc_show: adc 1544
[  371.715142] Governor: exit mode 0x0
[  371.718613] [SSP]: hub_chg_adc_show: adc 1193
[  371.727653] reading MSM_THERM_CH [rc = 0] [adc_code = 30080]
```

lavanoid, 2021-06-10:
[wifi.tar.gz](https://github.com/seemoo-lab/nexmon/files/6627540/wifi.tar.gz)

lavanoid, 2021-06-10:
```
/vendor/etc/wifi # strings bcmdhd_sta.bin | tail -1                                                                                                                                                                                                                                 
4339a0-roml/sdio-ag-pool-p2p-pno-pktfilter-keepalive-aoe-sr-mchan-proptxstatus-lpc-wl11u-betdls-autoabn-txbf-rcc-relmcast-fmc-wepso-ve-ccx-okc-clm_4335_ss-txpwr-noccxaka-mfp-ndoe Version: 6.37.32.48.8 (A0 Station/P2P) CRC: 5276f1 Date: Mon 2018-02-26 16:59:20 KST Ucode Ver: 855.10406 FWID: 01-942e430c
```

matthiasseemoo, 2021-06-10:
It is not too different from 6_37_34_43 regarding the version number. But maybe something important changed and it is not working with your latest driver, or some features are missing. So you need to patch your new firmware version.

lavanoid, 2021-06-10:
OK. Is there a wiki page or some documentation I can follow, to patch it?

lavanoid, 2021-06-10:
This is perhaps the wrong place to ask, but I'd also be willing to discuss making a patch for this privately and tipping for it.

lavanoid, 2021-06-11:
https://github.com/lavanoid/nexmon/commit/80ccab428c48e82fb6a67d6e6758a84cea479401

I've tried modifying the patches that currently exist for the Nexus 5, but I am unable to run the `dump-rom` command successfully.
[image]

lavanoid, 2021-06-11:
I cannot install `dhdutil` either. It's not pre-installed on Lineage, and the Nexmon app cannot install it due to Android 11 not allowing the /system partition to be mounted as rw. If I could at least install the utilities  and libraries inside of the apps data directory, I could probably do everything via adb root shell.

lavanoid, 2021-06-11:
OK. So I found an older version of Nexutil and extracted the utilities from it ( https://github.com/seemoo-lab/bcm-public/blob/master/androidstudioprojects/Nexmon/app/app-release.apk ) as the Nexutil app from the Nethunter store didn't seem to have any binaries inside of it (?).

I pushed `dhdutil` to /data, marked it as executable and ran `.dhdutil membytes -r 0x0 0xA0000 > rom.bin`

Here is the outputted ROM file: 
[rom.zip](https://github.com/seemoo-lab/nexmon/files/6641349/rom.zip)

lavanoid, 2021-06-12:
More info that might be useful. I'd super appreciate any help here. Considering the Nexus 5 firmware works fine, and it's just the Android WiFi stack that stops working, I wouldn't think it would be too much work for someone with an understanding.

```
95|hlte:/vendor/bin # dhdutil dump                                                                                                                                                                                                                                                       
Dongle Host Driver, version 1.141.64.59 (security_patch)
Compiled in drivers/net/wireless/bcmdhd on Jun  5 2021 at 08:11:23

pub.up 1 pub.txoff 0 pub.busstate 2
pub.hdrlen 48 pub.maxctl 10256 pub.rxsz 1738
pub.iswl 1 pub.drv_version 0 pub.mac f0:25:b7:12:a6:f5
pub.bcmerror -23 tickcnt 36600
dongle stats:
tx_packets 1351 tx_bytes 178680 tx_errors 0 tx_dropped 0
rx_packets 14031 rx_bytes 2730708 rx_errors 0 rx_dropped 0
multicast 7540
bus stats:
tx_packets 1351 tx_multicast 140 tx_errors 0
tx_ctlpkts 5034 tx_ctlerrs 0
rx_packets 14031 rx_multicast 7540 rx_errors 0 
rx_ctlpkts 5034 rx_ctlerrs 0 rx_dropped 0
rx_readahead_cnt 0 tx_realloc 0

Protocol CDC: reqid 5034

Bus SDIO structure:
hostintmask 0x000000f0 intstatus 0x00000000 sdpcm_ver 4
fcstate 0 qlen 0 tx_seq 130, max 162, rxskip 0 rxlen 0 rx_seq 201
intr 1 intrcount 15331 lastintrs 0 spurious 0
pollrate 0 pollcnt 0 regfails 0

Additional counters:
tx_tailpad_chain 43 tx_tailpad_pktget 0
tx_sderrs 0 fcqueued 1335 rxrtx 0 rx_toolong 0 rxc_errors 0
rx_hdrfail 0 badhdr 0 badseq 27
fc_rcvd 0, fc_xoff 0, fc_xon 0
rxglomfail 0, rxglomframes 686, rxglompkts 1559
f2rx (hdrs/data) 53450 (34536/18914), f2tx 6297 f1regs 30686

Rx: pkts/f2rd 0.26, pkts/f1sd 0.45, pkts/sd 0.16, pkts/int 0.91
Rx: glom pct 11.11, pkts/glom 2.27
Tx: pkts/f2wr 0.21, pkts/f1sd 0.04, pkts/sd 0.03, pkts/int 0.08
Total: pkts/f2rw 0.25, pkts/f1sd 0.50, pkts/sd 0.17, pkts/int 1.00

dpc_sched 0 host interrupt not pending
blocksize 512 roundup 512
clkstate 3 activity 0 idletime 5 idlecount 0 sleeping 1
```
Kernel source code: https://github.com/LineageOS/android_kernel_samsung_msm8974

lavanoid, 2021-06-15:
Sorry for the spam. I've been updating my repo with stuff that will help with getting this working on the Note 3.
https://github.com/lavanoid/nexmon

I believe I have the base project structure there. Now it's just a matter of finding what the addresses are for stuff?

Headedbranch225, 2024-01-07:
I am having the same problem now, can you tell me if you have a solution?

HOUDA555, 2026-03-19:
Hello, I have the same device and I can install and run the program. Nexmon Successfully, but the problem is I don't have a NetHunter kernel for Android 11 or 12. Can anyone help? I tried building a kernel but it didn't work.