seemoo-lab/nexmon #481: RaspberryPi 4B: not possible connect to an access point while monitoring

LuViKu, 2021-06-14:
In the description it is stated, that it should be possible to do both monitoring and connecting to an access point. For me this doesnt seem to work. When starting the mon0 interface I loose the connection to my access point. I modified the installation process like stated in [issue#469](https://github.com/seemoo-lab/nexmon/issues/469), downgrading the raspian kernel and headers. Did any of you get both interfaces running simultaneously?

ssssleep, 2021-11-17:
hello,guys.I meet the same problem,I encountered the same problem. After I turned on the monitor mode, I couldn't connect to the AP. Did you solve it? RIP4

BrendanxP, 2023-11-20:
Same issue here still. Tried fiddling with it a lot but haven't had both working at the same time. Wired Ethernet works though.
I am now running Ubuntu with Linux kernel 5.15.