seemoo-lab/nexmon #483: Error while building driver for kernel 5.4.40

AvinashNargund, 2021-06-22:
I get the following error while running `make` in `patches/bcm43455c0/7_45_206/nexmon/` on Raspberry Pi 4 with kernel version 5.4.40
`ERROR: "brcmu_dbg_hex_dump" [/home/pi/nexmon/patches/bcm43455c0/7_45_206/nexmon/brcmfmac_5.4.y-nexmon/brcmfmac.ko] undefined!
ERROR: "__brcmf_dbg" [/home/pi/nexmon/patches/bcm43455c0/7_45_206/nexmon/brcmfmac_5.4.y-nexmon/brcmfmac.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[1]: *** [Makefile:1607: modules] Error 2
make: *** [Makefile:46: brcmfmac.ko] Error 2
`

Any ideas on how I can resolve this?

jlinktu, 2021-06-22:
Since the kernel got updated to 5.10, following the readme installs the wrong kernel headers. You need those specific to your kernel version. A simple way to install them is using https://github.com/RPi-Distro/rpi-source

We plan on adding support for 5.10 soon

AvinashNargund, 2021-06-22:
I see, thanks for the info. But I think I picked the right kernel headers by following #338 
Are those headers different from the kernel headers you are referring to?

jlinktu, 2021-06-22:
I have not tested the approach in #338, but if you choose the correct headers according to your kernel version and architecture it might also work.

You can give above approach using rpi-source a try and see if it works, if it does not, the headers might not be the problem.

AvinashNargund, 2021-06-22:
Gotcha, will give that a go