seemoo-lab/nexmon #484: Monitor mode detects nothing on RPi CM4

krort, 2021-06-28:
I can't get monitor mode to work on my Raspberry Pi Compute Module 4:

```bash
sudo aireplay-ng --test mon0

09:35:59  Trying broadcast probe requests...
09:36:01  No Answer...
09:36:01  Found 0 APs
```

Some information:
RPi: Raspberry Pi Compute Module 4

```bash
hostnamectl

  ...
  Operating System: Raspbian GNU/Linux 10 (buster)
            Kernel: Linux 5.4.51-v7l+
      Architecture: arm
```

After turning on the RPi, I run the following commands:

```bash
sudo su
cd /home/pi/nexmon/
source setup_env.sh
cd patches/bcm43455c0/7_45_206/nexmon
make install-firmware
iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor
ifconfig mon0 up
```

But when i run `airodump-ng mon0 -c 6`, nothing happens:

```bash
airodump-ng mon0 -c 6

CH  6 ][ Elapsed: 0 s ][ 2021-06-23 09:31 ][ fixed channel mon0: -1

BSSID              PWR RXQ  Beacons    #Data, #/s  CH  MB   ENC  CIPHER AUTH ESSID

BSSID              STATION            PWR   Rate    Lost    Frames  Probe
```

```bash
airodump-ng mon0 -c 6 --ig

CH  6 ][ Elapsed: 21 s ][ 2021-06-23 09:32

BSSID              PWR RXQ  Beacons    #Data, #/s  CH  MB   ENC  CIPHER AUTH ESSID

BSSID              STATION            PWR   Rate    Lost    Frames  Probe
```

No error, but monitor mode detects nothing. It is the same for `airodump-ng mon0`.
When I run the same command on my computer (`sudo iw phy phy0 interface add mon0 type monitor; airmon-ng check kill; airodump-ng mon0 -c 6`), it detects several networks, so I should no be too far away to detect anything.
I have tried to wait for a longer period of time, but still nothing. I have also tried to use `airmon-ng check kill`, but it did not help.

Additional info:

```bash
iw dev

phy#1
	Interface mon0
		ifindex 6
		wdev 0x100000003
		addr dc:a6:32:fe:ec:3d
		type monitor
	Interface wlan0
		ifindex 5
		wdev 0x100000001
		addr dc:a6:32:fe:ec:3d
		type managed
```

```bash
ip addr show mon0

6: mon0: <BROADCAST,ALLMULTI,PROMISC,NOTRAILERS,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ieee802.11/radiotap dc:a6:32:fe:ec:19 brd ff:ff:ff:ff:ff:ff
```

```bash
sudo airmon-ng

PHY	Interface	Driver		Chipset

phy1	mon0		brcmfmac	Broadcom 43430
phy1	wlan0		brcmfmac	Broadcom 43430
```

```bash
sudo airmon-ng start mon0

PHY	Interface	Driver		Chipset

phy1	mon0		brcmfmac	Broadcom 43430

Error setting channel: command failed: Device or resource busy (-16)
Error -16 likely means your card was set back to station mode by something.
Removing non-monitor mon0 interface...

WARNING: unable to start monitor mode, please run "airmon-ng check kill"
```

I have already used `airmon-ng check kill`, so it returns nothing. Before I used it, it returned this:

```bash
sudo airmon-ng check

Found 5 processes that could cause trouble.
Kill them using 'airmon-ng check kill' before putting
the card in monitor mode, they will interfere by changing channels
and sometimes putting the interface back in managed mode

  PID Name
  387 avahi-daemon
  405 wpa_supplicant
  418 avahi-daemon
  565 dhcpcd
 3518 wpa_supplicant
```

```bash
sudo airmon-ng start wlan0

PHY	Interface	Driver		Chipset

phy1	mon0		brcmfmac	Broadcom 43430
phy1	wlan0		brcmfmac	Broadcom 43430

Error setting channel: command failed: Device or resource busy (-16)
Error -16 likely means your card was set back to station mode by something.
Removing non-monitor mon0 interface...

WARNING: unable to start monitor mode, please run "airmon-ng check kill"
```

```bash
iwconfig
eth0      no wireless extensions.

mon0      IEEE 802.11  Mode:Monitor
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on

eth1      no wireless extensions.

wlan0     IEEE 802.11  ESSID:off/any
          Mode:Managed  Frequency:2.437 GHz  Access Point: Not-Associated
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off

lo        no wireless extensions.
```

French21, 2021-07-30:
Hey mate, did you find a solution? Having the same issue on the S10

krort, 2021-08-02:
> Hey mate, did you find a solution? Having the same issue on the S10

No, I didn't. I changed to Ubuntu 20.04 and ad-hoc (ibss) mode instead. But if please tell me if you manage to do it!

lucidyan, 2022-07-31:
I've been in the same problem, and solved it by starting monitoring interface with `sudo airmon-ng start wlan0` instead of `iw`. Keep in mind that name of new interface will be `wlan0mon`, not `mon0`.
I hope this helps.