seemoo-lab/nexmon #488: nexmon crashing

cor33, 2021-07-15:
Im using a nexus 6p running android 8.1.0 rooted with magisk using the latest release of nethunter for oreo. The install went smooth unlockboot loader,flash twrp,custom rom magisk then nethunter after reboot after reboot lol. Now nexmon comes with new nethunter and works as far as i can tell via android su terminal but the nexmon that i downloaded from nethunter store it asks on the start up request root and check for files when i click it it just stops ,force closes and crashes, after it happend the first time i went in manually allowed it to have superuser rights and tried it agian ive tried older versions but just as the new release on nethunter store they didnt work ,please get this working id love to be able to use the nexmon app alongside of nethuntef on my nexus 6p oreo

atkn1337, 2021-10-14:
it happened to me im using nethunter os in nexus 5

cor33, 2021-10-14:
Any idea how to get n6p fully working with nexmon I've tried and tried and zero luck

Sent with [ProtonMail](https://protonmail.com/) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

e2002e, 2022-01-14:
Take this one https://github.com/e2002e/nexmon/releases/tag/notrack

t0int, 2022-01-26:
> Take this one https://github.com/e2002e/nexmon/releases/tag/notrack

Fixed the issue