seemoo-lab/nexmon #497: Debug options - printf?

tomer-avrahami, 2021-10-02:
Hi,

I am trying to debug my system\code.
How can I see printf output? Is there another recommended debug option?

Is there a recommended platform I should use? I am currently using eclipse, but any other IDE which has the project properties would be great.

Thanks,
Tomer

Alexxdal, 2021-11-16:
Up

jlinktu, 2025-03-18:
The equivalent to what @itsmattkc posted above for `brcmfmac` can be achieved for `bcmdhd` by setting the kernel module parameter `dhd_msg_level` to include `0x400000`.

A more general approach is to dump the firmware internal console (which is where `printf` outputs end up) by using an IOCTL. An example on how this can be done is here: https://github.com/seemoo-lab/nexmon/blob/ef25ce36700acf60f91ebc58c43f0b2d8dc372cd/patches/bcm4389c1/20_101_57_r1035009/nexmon/src/ioctl.c#L50

closed 2025-03-18 by jlinktu