seemoo-lab/nexmon #500: Does it work on rpi zero 2 W ?

stillpolar, 2021-11-11:
Pi zero 2 W has bcm2710a1 chip, I don't have it on hand but if anyone owns it can you test if nexmon works on the new raspberry pi zero 2 W too ?

matthiasseemoo, 2021-11-11:
you could run nexutil -V and post the output for a start

skontrolle, 2021-11-15:
Here's a `nextutil -V` dump on a zero2. 

```
firmware 9.88.4.65 (test) (f149b32@shgit)  (r679549) FWID 01-f40f3270
vendorid 0x14e4
deviceid 0x43e2
radiorev 0x403da000
chipnum 0xa9a6
chiprev 0x2
chippackage 0x4
corerev 0x27
boardid 0x726
boardvendor 0x14e4
boardrev P101
driverrev 0x9580000
ucoderev 0x4135105
bus 0x0
phytype 0xc
phyrev 0x1
anarev 0x0
nvramrev 0x0
```

matthiasseemoo, 2021-11-15:
> firmware 9.88.4.65 (test) ***@***.***)  (r679549) FWID 01-f40f3270

It looks like a new revision of the chip already known from the Raspberry Pi 3. See also: https://github.com/seemoo-lab/nexmon/blob/master/REVINFO.md <https://github.com/seemoo-lab/nexmon/blob/master/REVINFO.md>

So you could try, if its firmware also runs on the Zero 2 W. If not, you need to patch the Zero 2 W firmware.

skontrolle, 2021-11-16:
@matthiasseemoo thanks for the tip.  I was unable to get anything renaming the 7_45_41_46 firmware to 43436.  I think we might need to patch the 43436 firmware.  How involved is that?  Are the patches similar between generations?

Complaint when using renamed patched brcmfmac43430-sdio:
```
 brcmfmac: brcmf_sdio_readshared: sdpcm shared version unsupported: dhd 3 dongle 253
```

Successful modprobe with the [official firmware](https://github.com/RPi-Distro/firmware-nonfree/tree/master/brcm):
```
brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/2 wl0: Oct  9 2020 14:44:32 version 9.88.4.65 (test) (f149b32@shgit)  (r679549) FWID 01-f40f3270
```

MaybeAdmin, 2021-11-23:
Any updates on this?

isthisausername2, 2021-11-23:
Not that I know off. Still waiting on the driver updates.

DrSchottky, 2022-01-04:
Started looking at it, it's using a different format for flashpatches (looks like bcm43596 patches format)
Working on rom dump, any help would be appreciated

skontrolle, 2022-01-07:
How can I help?

DrSchottky, 2022-01-08:
I'd need someone experienced in rev eng helping me finding all the offsets/patches for this new fw.
I already ported definitions.mk and part of rom_extraction code, but even the simplest patch doesn't work (chip is dead after loading custom fw).
@matthiasseemoo is [patch.c](https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm43430a1/7_45_41_46/nexmon/src/patch.c) the bare minimum to test if definitions.mk is ok, right?

matthiasseemoo, 2022-01-08:
generally yes. it will ensure that ucode compression works so that you have enough space for other patches.

DrSchottky, 2022-01-08:
I forked project and committed my findings
https://github.com/DrSchottky/nexmon/tree/rpizero2w

If anyone wants to help me figuring out what's wrong let me know

DrSchottky, 2022-01-15:
Got monitor mode working :)

AlexWhitehouse, 2022-02-02:
Any update on this?

DrSchottky, 2022-02-02:
Yes, it's done.
I'm waiting for permission to publish it.

davenicoll, 2022-02-02:
> Yes, it's done.
> I'm waiting for permission to publish it.

Awesome job!

shakeTheC0de, 2022-02-02:
Plsss i need it

isthisausername2, 2022-02-03:
> Yes, it's done.
> I'm waiting for permission to publish it.

Very good work, we all thank your for your work and support.

davenicoll, 2022-02-03:
any ETA on this @DrSchottky ? (sorry, we're all waiting anxiously for the patched driver)

matglas, 2022-02-03:
Out of curiosity. Who needs to give permission for publication? And how does that work?

DrSchottky, 2022-02-03:
> Out of curiosity. Who needs to give permission for publication? And how does that work?

Who sponsored me to work on this. 

@davenicoll I hope in the next few weeks.

matglas, 2022-02-03:
> Who sponsored me to work on this.

Thank you!

DrSchottky, 2022-02-19:
I sent a PR with the patches
https://github.com/seemoo-lab/nexmon/pull/514

Tested it with basic patches (monitor mode and  injection) and they look fine. Your feedback is welcome

shakeTheC0de, 2022-02-19:
Ty I goto test it now!!

DrSchottky, 2022-02-20:
Found a bug: If you inject frames too fast or do chan hopping while injecting (both scenarios can be reproduced with aireplay) firmware  crashes
```
FWID 01-f40f3270
flags 1

TRAP c(6ff7c): pc 348a0, lr 349e5, sp 3f158, cpsr 2000000c, spsr 21000000
  r0 1a7c, r1 5fae4, r2 1a7c, r3 2011, r4 65574, r5 0, r6 5fa20
  r7 5fa9c, r8 65574, r9 f0, r10 0, r11 6a050, r12 0

   sp+0 0005d7c4 00065574 00000000 0006d864
  sp+10 0005faa0 00000001 0003f234 00000006

sp+34 0000584f
sp+b8 00034859
sp+cc 00031fad
sp+e0 0000ffff
sp+114 0000be03
sp+1ac 00001ec5
sp+1dc 00003f19
sp+214 00004283
sp+22c 00004575
sp+23c 00004953
sp+244 0081f351
sp+254 00004dc7
sp+25c 00004bc1
sp+274 008251a5
sp+410 0000800f
```
Working on a patch...

blackfunnyduck, 2022-02-21:
I tried to use the code from https://github.com/DrSchottky/nexmon.git.
However when compiling the patched firmware for bcm43436b0 I got this error message:

BUILDING DRIVER for kernel 5.4 brcmfmac_5.4.y-nexmon/brcmfmac.ko (details: log/driver.log)
scripts/Makefile.build:42: /home/kali/nexmon/patches/bcm43436b0/9_88_4_65/nexmon/brcmfmac_5.4.y-nexmon/Makefile: No such file or directory
make[2]: *** No rule to make target '/home/daffy/nexmon/patches/bcm43436b0/9_88_4_65/nexmon/brcmfmac_5.4.y-nexmon/Makefile'.  Stop.
make[1]: *** [Makefile:1732: /home/daffy/nexmon/patches/bcm43436b0/9_88_4_65/nexmon/brcmfmac_5.4.y-nexmon] Error 2
make: *** [Makefile:46: brcmfmac.ko] Error 2

Actually the brcmfmac_5.4.y-nexmon folder is void.
My pi zero 2 w is running Kali, kernel version 5.4.83-Re4son-v7+
Your support would be much appreciated.

moni11811, 2022-02-21:
Actually the brcmfmac_5.4.y-nexmon folder is void. My pi zero 2 w is running Kali, kernel version 5.4.83-Re4son-v7+

You need to update your kernel to 5.10 otherwise it doesnt install it and some libraries are not present in 5.10

If you are trying to run Pwnagotchi please check [#1046 in /pwnagotchi](https://github.com/seemoo-lab/nexmon/issues/500#issuecomment-1046295728)

DrSchottky, 2022-02-21:
I included brcmfmac only for 5.10 since it's the kernel shipped with the latest Raspberry Pi OS.
To use it on 5.4 build brcmfmac driver from bcm43455c0 or copy `bcm43455c0/7_45_206/nexmon/brcmfmac_5.4.y-nexmon` to `bcm43436b0/9_88_4_65/nexmon`

blackfunnyduck, 2022-02-21:
> bcm43436b0/9_88_4_65/nexmon

Thank you for your swift feedback.
I tried both solutions you suggested but unfortunately none worked.
When compiling the firmware the assembler returns a lot of error messages like these:

BUILDING DRIVER for kernel 5.4 brcmfmac_5.4.y-nexmon/brcmfmac.ko (details: log/driver.log)
/tmp/ccPxhgAK.s: Assembler messages:
/tmp/ccPxhgAK.s:3183: Error: selected processor does not support `dmb ish' in ARM mode
/tmp/ccPxhgAK.s:3192: Error: selected processor does not support `dsb ishst' in ARM mode
/tmp/ccPxhgAK.s:3195: Error: selected processor does not support `sev' in ARM mode
/tmp/ccPxhgAK.s:3211: Error: selected processor does not support `dmb ish' in ARM mode
/tmp/ccPxhgAK.s:3220: Error: selected processor does not support `dsb ishst' in ARM mode
/tmp/ccPxhgAK.s:3223: Error: selected processor does not support `sev' in ARM mode
/tmp/ccPxhgAK.s:3237: Error: selected processor does not support `dmb ish' in ARM mode
/tmp/ccPxhgAK.s:3246: Error: selected processor does not support `dsb ishst' in ARM mode
/tmp/ccPxhgAK.s:3249: Error: selected processor does not support `sev' in ARM mode
...
Maybe this is because I am trying to install the patches on Kali but these have been written for Raspbian / different kernel.
Any suggestion would be very welcome.

stillpolar, 2022-02-21:
I had some similar issues but I switched kernel

DrSchottky, 2022-02-21:
> Thank you for your swift feedback. I tried both solutions you suggested but unfortunately none worked. When compiling the firmware the assembler returns a lot of error messages like these:
>
> BUILDING [...]

`/usr/bin/gcc --version`
if > 8.X downgrade to gcc-8

ShadyNawara, 2022-02-22:
I am getting  `ERROR: unable to inject packet (send: Resource temporarily unavailable)` when running [owl](https://github.com/seemoo-lab/owl) could this be related to the latest patch?

blackfunnyduck, 2022-03-02:
> > > bcm43436b0/9_88_4_65/nexmon
> > 
> > 
> > Thank you for your swift feedback. I tried both solutions you suggested but unfortunately none worked. When compiling the firmware the assembler returns a lot of error messages like these:
> > BUILDING DRIVER for kernel 5.4 brcmfmac_5.4.y-nexmon/brcmfmac.ko (details: log/driver.log) /tmp/ccPxhgAK.s: Assembler messages: /tmp/ccPxhgAK.s:3183: Error: selected processor does not support `dmb ish' in ARM mode /tmp/ccPxhgAK.s:3192: Error: selected processor does not support `dsb ishst' in ARM mode /tmp/ccPxhgAK.s:3195: Error: selected processor does not support `sev' in ARM mode /tmp/ccPxhgAK.s:3211: Error: selected processor does not support `dmb ish' in ARM mode /tmp/ccPxhgAK.s:3220: Error: selected processor does not support `dsb ishst' in ARM mode /tmp/ccPxhgAK.s:3223: Error: selected processor does not support `sev' in ARM mode /tmp/ccPxhgAK.s:3237: Error: selected processor does not support `dmb ish' in ARM mode /tmp/ccPxhgAK.s:3246: Error: selected processor does not support `dsb ishst' in ARM mode /tmp/ccPxhgAK.s:3249: Error: selected processor does not support `sev' in ARM mode ... Maybe this is because I am trying to install the patches on Kali but these have been written for Raspbian / different kernel. Any suggestion would be very welcome.
> 
> `/usr/bin/gcc --version` if > 8.X downgrade to gcc-8
> 
> Or switch to Raspberry Pi OS

I installed a fresh Raspberry pi OS lite but still is not working.
When running 'source setup_env.sh' it returns the "Platform not supported!" error.
Here are the kernel and processor details:
 5.10.92-v8+ aarch64 GNU/Linux

processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : 902120
Serial          : 00000000eef78b2d
Model           : Raspberry Pi Zero 2 W Rev 1.0

Any ideas and suggestions would be very welcome.

DrSchottky, 2022-03-02:
Use 32bit Raspberry Pi OS

moni11811, 2022-03-03:
I am running into these issues on pwnagotchi
This seems to be a nexmon issue as other users aren't experiencing these issues with external wifi cards...

`[ERROR]    Task exception was never retrieved
future: <Task finished coro=<Client.start_websocket() done, defined at /usr/local/lib/python3.7/dist-packages/pwnagotchi/bettercap.py:38> exception=OSError("Multiple exceptions: [Errno 111] Connect call failed ('::1', 8081, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 8081)")>
OSError: Multiple exceptions: [Errno 111] Connect call failed ('::1', 8081, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 8081)
and 
[ERROR]    Error while setting wifi.recon.channels (HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x43540610>: Failed to establish a new connection: [Errno 111] Connection refused')))
ConnectionRefusedError: [Errno 111] Connection refused
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object at 0x43540610>: Failed to establish a new connection: [Errno 111] Connection refused
raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x43540610>: Failed to establish a new connection: [Errno 111] Connection refused'))
raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x43540610>: Failed to establish a new connection: [Errno 111] Connection refused'))`

My best guess is ephemeral port exhaustion

blackfunnyduck, 2022-03-03:
> Use 32bit Raspberry Pi OS

Thank you for your support.
I used a 32 bit Raspberry Pi OS and this time the patch compiling process and installation worked with no error.
However, running "iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor" returns "command failed: Operation not supported (-95)".
This may indicate the patch has failed to install?
Also, I ran airmon-ng:
PHY     Interface       Driver          Chipset
phy0    wlan0           brcmfmac    Broadcom 43430
This also does not look right to me.
I am a bit confused with this so your kind assistance would be much appreciated.

moni11811, 2022-03-03:
> Thank you for your support. I used a 32 bit Raspberry Pi OS and this time the patch compiling process and installation worked with no error. However, running "iw phy `iw dev wlan0 info | gawk [...]

After installing did you move the driver as explained here:

> Optional: To make the RPI3 load the modified driver after reboot:
Find the path of the default driver at reboot: modinfo brcmfmac #the first line should be the full path
Backup the original driver: mv "<PATH TO THE DRIVER>/brcmfmac.ko" "<PATH TO THE DRIVER>/brcmfmac.ko.orig"
Copy the modified driver (Kernel 4.9): cp /home/pi/nexmon/patches/bcm43430a1/7_45_41_46/nexmon/brcmfmac_kernel49/brcmfmac.ko "<PATH TO THE DRIVER>/"
Copy the modified driver (Kernel 4.14): cp /home/pi/nexmon/patches/bcm43430a1/7_45_41_46/nexmon/brcmfmac_4.14.y-nexmon/brcmfmac.ko "<PATH TO THE DRIVER>/"
Probe all modules and generate new dependency: depmod -a
The new driver should be loaded by default after reboot: reboot  * Note: It is possible to connect to an access point or run your own access point in parallel to the monitor mode interface on the wlan0 interface.

blackfunnyduck, 2022-03-03:
l

> > > Use 32bit Raspberry Pi OS
> > 
> > 
> > Thank you for your support. I used a 32 bit Raspberry Pi OS and this time the patch compiling process and installation worked with no error. However, running "iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor" returns "command failed: Operation not supported (-95)". This may indicate the patch has failed to install? Also, I ran airmon-ng: PHY Interface Driver Chipset phy0 wlan0 brcmfmac Broadcom 43430 This also does not look right to me. I am a bit confused with this so your kind assistance would be much appreciated.
> 
> After installing did you move the driver as explained here:
> 
> > Optional: To make the RPI3 load the modified driver after reboot:
> > Find the path of the default driver at reboot: modinfo brcmfmac #the first line should be the full path
> > Backup the original driver: mv "/brcmfmac.ko" "/brcmfmac.ko.orig"
> > Copy the modified driver (Kernel 4.9): cp /home/pi/nexmon/patches/bcm43430a1/7_45_41_46/nexmon/brcmfmac_kernel49/brcmfmac.ko "/"
> > Copy the modified driver (Kernel 4.14): cp /home/pi/nexmon/patches/bcm43430a1/7_45_41_46/nexmon/brcmfmac_4.14.y-nexmon/brcmfmac.ko "/"
> > Probe all modules and generate new dependency: depmod -a
> > The new driver should be loaded by default after reboot: reboot  * Note: It is possible to connect to an access point or run your own access point in parallel to the monitor mode interface on the wlan0 interface.

Thank you so much for the heads-up.
Indeed I missed this crucial step, my bad.
I copied the patched driver to /lib/modules/5.10.92-v7+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko and the mon0 has been created. Apparently this up and running but it doesn't capture any traffic.
tcpdump -i mon0 returns:
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on mon0, link-type IEEE802_11_RADIO (802.11 plus radiotap header), snapshot length 262144 bytes
^C
0 packets captured
0 packets received by filter
0 packets dropped by kernel
Also, airodump-ng cannot capture traffic with mon0
Here is what the system reports on mon0:
ifconfig:
mon0: flags=867<UP,BROADCAST,NOTRAILERS,RUNNING,PROMISC,ALLMULTI>  mtu 1500
        unspec E4-5F-01-7D-CC-95-08-50-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.86.182  netmask 255.255.255.0  broadcast 192.168.86.255
        inet6 fe80::a427:e67b:19f9:194f  prefixlen 64  scopeid 0x20<link>
        ether e4:5f:01:7d:cc:95  txqueuelen 1000  (Ethernet)
        RX packets 375891  bytes 69764393 (66.5 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3687  bytes 667910 (652.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

iwconfig:
wlan0     IEEE 802.11  ESSID:"homersi"
          Mode:Managed  Frequency:2.437 GHz  Access Point: 70:3A:CB:83:EC:B1
          Bit Rate=65 Mb/s   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          Link Quality=70/70  Signal level=-40 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:15  Invalid misc:0   Missed beacon:0

mon0      IEEE 802.11  Mode:Monitor  Frequency:2.437 GHz  Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on

ip a:
2: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether e4:5f:01:7d:cc:95 brd ff:ff:ff:ff:ff:ff
    inet 192.168.86.182/24 brd 192.168.86.255 scope global dynamic noprefixroute wlan0
       valid_lft 82785sec preferred_lft 71985sec
    inet6 fe80::a427:e67b:19f9:194f/64 scope link
       valid_lft forever preferred_lft forever
4: mon0: <BROADCAST,ALLMULTI,PROMISC,NOTRAILERS,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ieee802.11/radiotap e4:5f:01:7d:cc:95 brd ff:ff:ff:ff:ff:ff

iw dev:
 Interface mon0
                ifindex 4
                wdev 0x4
                addr e4:5f:01:7d:cc:95
                type monitor
                channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
                txpower 31.00 dBm
        Unnamed/non-netdev interface
                wdev 0x2
                addr 22:a0:89:e0:c7:f1
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 2
                wdev 0x1
                addr e4:5f:01:7d:cc:95
                ssid homersi
                type managed
                channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
                txpower 31.00 dBm

System: Raspberry pi zero 2 w, Raspberry Pi OS, kernel  5.10.92-v7+
My experience with patched drivers is quite limited so your further suggestions would be much appreciated.

moni11811, 2022-03-03:
> l
> 
> > > > Use 32bit Raspberry Pi OS
> > > 
> > > 
> > > Thank you for your support. I used a 32 bit Raspberry Pi OS and this time the patch compiling process and installation worked with no error. However, running "iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor" returns "command failed: Operation not supported (-95)". This may indicate the patch has failed to install? Also, I ran airmon-ng: PHY Interface Driver Chipset phy0 wlan0 brcmfmac Broadcom 43430 This also does not look right to me. I am a bit confused with this so your kind assistance would be much appreciated.
> > 
> > 
> > After installing did you move the driver as explained here:
> > > Optional: To make the RPI3 load the modified driver after reboot:
> > > Find the path of the default driver at reboot: modinfo brcmfmac #the first line should be the full path
> > > Backup the original driver: mv "/brcmfmac.ko" "/brcmfmac.ko.orig"
> > > Copy the modified driver (Kernel 4.9): cp /home/pi/nexmon/patches/bcm43430a1/7_45_41_46/nexmon/brcmfmac_kernel49/brcmfmac.ko "/"
> > > Copy the modified driver (Kernel 4.14): cp /home/pi/nexmon/patches/bcm43430a1/7_45_41_46/nexmon/brcmfmac_4.14.y-nexmon/brcmfmac.ko "/"
> > > Probe all modules and generate new dependency: depmod -a
> > > The new driver should be loaded by default after reboot: reboot  * Note: It is possible to connect to an access point or run your own access point in parallel to the monitor mode interface on the wlan0 interface.
> 
> Thank you so much for the heads-up. Indeed I missed this crucial step, my bad. I copied the patched driver to /lib/modules/5.10.92-v7+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko and the mon0 has been created. Apparently this up and running but it doesn't capture any traffic. tcpdump -i mon0 returns: tcpdump: verbose output suppressed, use -v[v]... for full protocol decode listening on mon0, link-type IEEE802_11_RADIO (802.11 plus radiotap header), snapshot length 262144 bytes ^C 0 packets captured 0 packets received by filter 0 packets dropped by kernel Also, airodump-ng cannot capture traffic with mon0 Here is what the system reports on mon0: ifconfig: mon0: flags=867<UP,BROADCAST,NOTRAILERS,RUNNING,PROMISC,ALLMULTI> mtu 1500 unspec E4-5F-01-7D-CC-95-08-50-00-00-00-00-00-00-00-00 txqueuelen 1000 (UNSPEC) RX packets 0 bytes 0 (0.0 B) RX errors 0 dropped 0 overruns 0 frame 0 TX packets 0 bytes 0 (0.0 B) TX errors 0 dropped 0 overruns 0 carrier 0 collisions 0
> 
> wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST> mtu 1500 inet 192.168.86.182 netmask 255.255.255.0 broadcast 192.168.86.255 inet6 fe80::a427:e67b:19f9:194f prefixlen 64 scopeid 0x20 ether e4:5f:01:7d:cc:95 txqueuelen 1000 (Ethernet) RX packets 375891 bytes 69764393 (66.5 MiB) RX errors 0 dropped 0 overruns 0 frame 0 TX packets 3687 bytes 667910 (652.2 KiB) TX errors 0 dropped 0 overruns 0 carrier 0 collisions 0
> 
> iwconfig: wlan0 IEEE 802.11 ESSID:"homersi" Mode:Managed Frequency:2.437 GHz Access Point: 70:3A:CB:83:EC:B1 Bit Rate=65 Mb/s Tx-Power=31 dBm Retry short limit:7 RTS thr:off Fragment thr:off Encryption key:off Power Management:on Link Quality=70/70 Signal level=-40 dBm Rx invalid nwid:0 Rx invalid crypt:0 Rx invalid frag:0 Tx excessive retries:15 Invalid misc:0 Missed beacon:0
> 
> mon0 IEEE 802.11 Mode:Monitor Frequency:2.437 GHz Tx-Power=31 dBm Retry short limit:7 RTS thr:off Fragment thr:off Power Management:on
> 
> ip a: 2: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000 link/ether e4:5f:01:7d:cc:95 brd ff:ff:ff:ff:ff:ff inet 192.168.86.182/24 brd 192.168.86.255 scope global dynamic noprefixroute wlan0 valid_lft 82785sec preferred_lft 71985sec inet6 fe80::a427:e67b:19f9:194f/64 scope link valid_lft forever preferred_lft forever 4: mon0: <BROADCAST,ALLMULTI,PROMISC,NOTRAILERS,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000 link/ieee802.11/radiotap e4:5f:01:7d:cc:95 brd ff:ff:ff:ff:ff:ff
> 
> iw dev: Interface mon0 ifindex 4 wdev 0x4 addr e4:5f:01:7d:cc:95 type monitor channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz txpower 31.00 dBm Unnamed/non-netdev interface wdev 0x2 addr 22:a0:89:e0:c7:f1 type P2P-device txpower 31.00 dBm Interface wlan0 ifindex 2 wdev 0x1 addr e4:5f:01:7d:cc:95 ssid homersi type managed channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz txpower 31.00 dBm
> 
> System: Raspberry pi zero 2 w, Raspberry Pi OS, kernel 5.10.92-v7+ My experience with patched drivers is quite limited so your further suggestions would be much appreciated.

That's because wlan0 seems to be connected to your home network.
Mon0 needs exclusive access to it to work
And wlan0 should be showing in an status Unknown
Either disconect and reboot, or uninstall wpa_supplicant

blackfunnyduck, 2022-03-03:
> * Note: It is possible to connect to an access point or run your own access point in parallel to the monitor mode interface on the wlan0 interface

Thank you for your reply.
"* Note: It is possible to connect to an access point or run your own access point in parallel to the monitor mode interface on the wlan0 interface"
This is actually not possible to stay connected to an AP and run mon0 in parallel? This is confusing.
Disconnecting the wlan0 from the AP will make impossible to control the headless pi.
I am sorry for bothering you with these questions but my understanding was that it is possible to run mon0 in monitoring mode while keeping the wlan0 connected to a network, i.e. by attaching two devs (wlan0 and mon0) to a single phy0. Maybe I've got it all wrong? 
Thank you for your patience.

blackfunnyduck, 2022-03-03:
> > > * Note: It is possible to connect to an access point or run your own access point in parallel to the monitor mode interface on the wlan0 interface
> > 
> > 
> > Thank you for your reply. "* Note: It is possible to connect to an access point or run your own access point in parallel to the monitor mode interface on the wlan0 interface" This is actually not possible to stay connected to an AP and run mon0 in parallel? This is confusing. Disconnecting the wlan0 from the AP will make impossible to control the headless pi. I am sorry for bothering you with these questions but my understanding was that it is possible to run mon0 in monitoring mode while keeping the wlan0 connected to a network, i.e. by attaching two devs (wlan0 and mon0) to a single phy0. Maybe I've got it all wrong? Thank you for your patience.
> 
> As @moni11811 stated, it's not possible to use the wlan adapter and have it in monitor mode, no. If you need access to your pi, connect it to a laptop via USB and SSH from there (how to guide: https://pwnagotchi.ai/configuration/#connecting-to-pi0w-with-microusb-cable-on-linux-host).

Thank you for clarifying this topic.
Then the note saying "It is possible to connect to an access point or run your own access point in parallel to the monitor mode interface on the wlan0 interface" from the README.md file at https://github.com/seemoo-lab/nexmon should be removed as this is a highly misleading statement.

closed 2022-03-05 by matthiasseemoo

R2Sam, 2022-04-28:
> That's because wlan0 seems to be connected to your home network. Mon0 needs exclusive access to it to work And wlan0 should be showing in an status Unknown Either disconect and reboot, or uninstall [...]

From my very poor understanding I have seemingly done all steps pointed out in the guide and what this thread has had to say and yet my wlan0 remains managed and my mon0 empty, would anyone be able to help. I'm happy to provide more details if are needed.

favdistro, 2025-01-26:
@DrSchottky in raspberry pi zero 2 w monitor mode working but clients not showing I'm currently using pi tail 2024.1

d4rks1d33, 2025-03-22:
> @DrSchottky in raspberry pi zero 2 w monitor mode working but clients not showing I'm currently using pi tail 2024.1, there is any solution to get working in monitor mode?

Hi, did you find a solution? I'm having the same problem

favdistro, 2025-03-22:
@d4rks1d33 yes try this https://github.com/rudyrdx/Raspberry-Pi-Zero-2w/blob/main/Monitor%20Mode.md

d4rks1d33, 2025-04-08:
@favdistro That tutorial didn't work for me netheir did the suggested solution on this thread. However I managed to get it working by deleting ALL the drivers in "/lib/firmware/brcm/" and replacing them with the ones inside a pwnagotchi . img for RPI zero 2 w (which did work in monitor mode) I also had to copy the "/lib/firmware/cypress" directory and make the corresponding symbolic links, when I have more time I'll make a tutorial with screenshots and the codes I used in case anyone else needs it (: 

PS: Monitor mode with packet injection works as long as I have a network connection (to activate it, just run airmon-ng start wlan0 and it will create wlan0mon in monitor mode while maintaining internet access)

[image]

Sonys9, 2025-04-11:
> [@favdistro](https://github.com/favdistro) That tutorial didn't work for me netheir did the suggested solution on this thread. However I managed to get it working by deleting ALL the drivers in [...]

[brcm_res.zip](https://github.com/user-attachments/files/19712916/brcm_res.zip)
BCRM drivers from pwnagotchi

Sonys9, 2025-04-11:
THATS WORKED!
Thank you, @d4rks1d33 

command to use for fix (patch nexmon before):
1. wget https://github.com/user-attachments/files/19712916/brcm_res.zip 
2. rm -rf /lib/firmware/brcm
3. unzip brcm_res.zip -d /lib/firmware/brcm

favdistro, 2025-04-16:
@d4rks1d33 bro please make full tutorial

favdistro, 2025-04-28:
@d4rks1d33 monitor mode working in pitail😱? how u get that working please make a full tutorial

favdistro, 2025-04-28:
@Sonys9 bro which u using rpos or pitail or please make tutorial on monitor mode

Sonys9, 2025-04-28:
@favdistro latest Raspberry OS
just patch with nexmon, then use my commands

favdistro, 2025-04-28:
@Sonys9 what about pitail how to get monitor mode working in pitail u know?

Sonys9, 2025-04-29:
@favdistro i dont use it, but you can do everything like on default raspberry pi OS

d4rks1d33, 2025-05-03:
@favdistro sorry for the delay I was busy working lol here is the tutorial:

https://github.com/d4rks1d33/Kali-Pi-Tail-with-monitor-mode-for-Raspberry-pi-zero-2-W/blob/main/README.md

As far as I know this only works on Kali PI Tail (I haven't tested it with any other OS) I plan to get pwnagotchi working in the future since this implementation supports the monitor mode needed for pwnagotchi to work and the managed mode so it can still have internet access, but it's a WIP and I'm 80% done with getting it working.

favdistro, 2025-05-03:
@d4rks1d33 Thanks broo. Packet injection working or not?

d4rks1d33, 2025-05-03:
@favdistro yes packet injection works perfect

favdistro, 2025-05-03:
@d4rks1d33 Thanks for reply