seemoo-lab/nexmon #501: RPI4 7_45_206 chanspec no 80MHz

marcospanghero, 2021-11-15:
Hi, I am trying to use frame injection on RPI4 with 7_45_206 and 5.4.y kernel.
I can compile the firmware and properly run it but seems that nexutil cannot properly change the channel to 36/80. It works without changing the bandwidth, moving the channel to 36, but other channels like 157 do not work.

Any suggestion? I am trying to generate packets for CSI measurements using nexmon_csi

marcospanghero, 2021-11-15:
From dmesg
`[ 4054.247370] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: calling brcmf_fil_cmd_data_set, cmd: 263
[ 4054.252045] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: Exit
`

matthiasseemoo, 2021-11-15:
you might need to configure this channel in your regulations.c file.

marcospanghero, 2021-11-15:
I will try that. On the other hands, why does `nexutil -k36/80` fail on 7_45_206?

matthiasseemoo, 2021-11-15:
Generally, you cannot directly switch to 80 MHz wide channels. However, if you add them in the regulations file, you can switch to any physically possible channel. Lets say even channel 7 in 2.4 GHz is possible with 80 MHz bandwidth.

marcospanghero, 2021-11-15:
I understand. So, should I patch regulations.c in nexmon_csi to support ch 36 or should I patch nexmon to support ch 36/80?

matthiasseemoo, 2021-11-15:
It looks like 36/80 is already configured in our nexmon csi code: https://github.com/seemoo-lab/nexmon_csi/blob/master/src/regulations.c#L218 <https://github.com/seemoo-lab/nexmon_csi/blob/master/src/regulations.c#L218>
So you might just need it for your transmitting firmware. You could also use the csi firmware for injection. You just need to add the code for frame injection.

marcospanghero, 2021-11-15:
could I just copy over injection.c from the 7_45_206 firmware and recompile? btw, where could I find an example of injection?

matthiasseemoo, 2021-11-15:
Copy also the sendframe.c.

Here, you find some examples: https://github.com/seemoo-lab/nexmon_csi/issues/251#issuecomment-962067658

marcospanghero, 2021-11-15:
thanks a lot!