seemoo-lab/nexmon #505: Using Nexmon with Pi 3b+ for Injection Frame

Hessenplay, 2022-01-07:
Hi there,
I would like the Pi 3b + for opendrop. For this I need the monitoring mode and injection frame. The list provided in the readme made me aware of the firmware patch 7_45_206 for the bcm43455c0. I'm trying to install this package, but don't know where to get the Raspberry Pi OS Kernel 5.4 for download. Does anyone have a download?
Thanks for the help in advance
Max

Jaqqon, 2022-01-09:
Hi!

Thanks to  "mildsunrise", monitoring and injection also works with the latest PiOS release (Oct 30, with the 5.10 kernel).

Please find the installation instructions specific to this release, here: [pios-wifi-injection.txt](https://github.com/seemoo-lab/nexmon/files/7835439/pios-wifi-injection.txt)

I have also created a simple injection example ([inject.c.txt](https://github.com/seemoo-lab/nexmon/files/7835443/inject.c.txt)).

Please note, that if you need to pause at least two milliseconds in between injections or otherwise the wifi firmware will most likely crash.

best regards

Bernhard

Jaqqon, 2022-01-09:
BTW.: the above installation instructions intentionally skipped the software update to make sure that the instructions work for everybody. However, I noticed that it takes only a couple of consecutive injections to crash the firmware. After software update and rebuilding the firmware / kernel driver, the continuous injection lasts for a couple of seconds before it crashes.

Here are the instructions for the software update (as of today).

A) Software update

sudo apt-get update
sudo apt full-upgrade
sudo reboot

B) Rebuild firmware and driver

cd nexmon
. setup_env.sh 
cd patches/bcm43455c0/7_45_206/nexmon
make clean && make
sudo cp brcmfmac43455-sdio.bin /lib/firmware/cypress/cyfmac43455-sdio.bin
sudo cp brcmfmac_5.10.y-nexmon/brcmfmac.ko /lib/modules/5.10.63-v7+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac

Hessenplay, 2022-01-17:
When I do make clean && make i get this Error. 
`make[1]: *** /lib/modules/5.10.89-v7+/build: No such file or directory.  Stop.
make: *** [Makefile:211: clean] Error 2
`
Where can I find the Logs?

Jaqqon, 2022-01-17:
Can you please provide information concerning how to install the 5.10.89 kernel so that I can reproduce the issue?

I just tried out the above instructions, again (including "Update Software" during the raspberry pi initial setup), but there's still the 5.10.63 kernel. And the injection works (with the known limits).

Jaqqon, 2022-01-17:
BTW.: "build" should be a symbolic link to the kernel build directory, i.e. in my case it looks like this:

> lrwxrwxrwx  1 root root     35 Dec  4 17:31 build -> /usr/src/linux-headers-5.10.63-v7l+

To be precise: in my case, this directory has the same structure as the kernel sources, but there are no *.c files included, just the header files.

YeetingWaterbottle, 2022-06-23:
> When I do make clean && make i get this Error. 
> `make[1]: *** /lib/modules/5.10.89-v7+/build: No such file or directory.  Stop.
> make: *** [Makefile:211: clean] Error 2
> `
> Where can I find the Logs?

I don't exactly know how to solve this problem, I ran into the same problem aswell but I think you need to install "raspberrypi os kernel header 5.10.89" after install that should create a folder that it's looking for.