seemoo-lab/nexmon #511: Is nexmon compatible with SYN43436?

blackfunnyduck, 2022-02-08:
I'm working on a raspberry pi zero 2 w project. The 2021 pi zero 2 w model seems to be fitted with SYN43436 built-in wifi chip.
The pi is running kali pi-tail edition (2022.1) that is coming with nexmon drivers.
Although the mon0 virtual interface is created and set in monitor mode, it doesn't seem to work. None of the traffic capture apps I tried returned any result as the mon0 does not detect any wifi devices.
Is the nexmon compatible with SYN43436?
My project should be based on the internal wifi interface and a USB attached wifi adapter must be avoided.
I would appreciate your comments and suggestions on how to make this work.

DrSchottky, 2022-02-08:
See https://github.com/seemoo-lab/nexmon/issues/500

blackfunnyduck, 2022-02-08:
Thank you for pointing me to #500 , as far as I understand  you are still waiting for your sponsors to approve the publishing of your code. Is this correct?

DrSchottky, 2022-02-08:
Exactly

wpa-2, 2023-03-18:
> Exactly

Any updates on this. 
Cheers

DrSchottky, 2023-03-18:
> > Exactly
> 
> Any updates on this. Cheers

It has been merged long ago