seemoo-lab/nexmon #516: wlan0 disappears after installing patches

IronBit-0, 2022-02-28:
I followed the instructions from the README on building and installing patches for the RPI3B. I've been running into an issue where after I install the firmware patches my wlan0 device vanishes.

By "install the firmware patches" I mean getting to the point in the instructions where you run `cd patches/bcm43430a1/7_45_41_46/nexmon/`, `make`, `make backup-firmware`, and then `make install-firmware`

output from `iw dev` before installing firmware patches
```
root@raspberrypi:/home/pi/nexmon# iw dev
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr ba:27:eb:6c:7d:69
                type P2P-device
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr b8:27:eb:6c:7d:69
                type managed
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
```

output from `iw dev` after installing firmware patches

```
root@raspberrypi:/home/pi/nexmon/patches/bcm43430a1/7_45_41_46/nexmon# iw dev
root@raspberrypi:/home/pi/nexmon/patches/bcm43430a1/7_45_41_46/nexmon#
```

I'm not sure if this repo is even still being updated, but I'll accept any ideas or questions.

Liuwe1Ye, 2022-04-18:
Have you solved this problem？I have the same one on nexus6p

IronBit-0, 2022-04-18:
On Raspberry Pi, I found that running `rpi-update 993f47507f287f5da56495f718c2d0cd05ccbc19` to change the kernel version to 4.19.97-v7+ (viewable by running `uname -r`) worked in some ways, but introduced other problems too.

I was trying to install the OpenDrop, which requires nexmon. I was running raspbian lite 32 bit. [This article](https://pimylifeup.com/raspberry-pi-nexmon/) was really helpful. Here are my log/note files, if they are any use:

[opendrop_commands_1.txt](https://github.com/seemoo-lab/nexmon/files/8506246/opendrop_commands_1.txt)
[opendrop_commands_2.txt](https://github.com/seemoo-lab/nexmon/files/8506247/opendrop_commands_2.txt)
[opendrop_commands_3.txt](https://github.com/seemoo-lab/nexmon/files/8506248/opendrop_commands_3.txt)
[opendrop_commands_4.txt](https://github.com/seemoo-lab/nexmon/files/8506249/opendrop_commands_4.txt)

I haven't worked on this project in a while, maybe I'll come back to it. Good luck.