seemoo-lab/nexmon #517: Can't find docs about how to patch

yesimxev, 2022-03-05:
I'm looking to patch TicWatch Pro's bcm43436b0 chipset fw version 9.88.0.0. As far as I see, it could be easily supported.

Is there any guide on how to edit the patches? So far I've got the fw_bcmdhd.bin but unsure of how to figure out patches just by looking at the other's patches.

kimocoder, 2022-03-05:
@matthiasseemoo

DrSchottky, 2022-03-07:
Support for bcm43436b0 (brcmfmac) has been added recently. 
If you're familiar with the tools I'd try to disassemble and bindiff brcmfmac/bcmdhd firmwares

yesimxev, 2022-03-08:
I'll surely do in the next few days. Any particular things to look for in the diffs?

kimocoder, 2022-03-08:
@DrSchottky I know your work from the best of several projects, may you enter the kerneltalk thinktank for a brief discussion? write me at Telegram or christian at aircrack-ng.org

shandongtlb, 2022-04-17:
> How awesome is that! I'll surely do in the next few days. Any particular things to look for in the diffs?

Hello, what's the latest progress? I've been studying ticwatch Pro3 LTE recently. The chip seems to be bcm43436b0, and the Linux kernel version is 4.9. Are you interested in working together

yesimxev, 2022-04-17:
Definitely! I also have the 3. I didn't have time to bindiff the first but both should be relatively easy. I have only little time so I postponed but I have some time here snd there 😃 Bith chipsets are supported and both arm. Imagine the Hijacker app on your watch 🤯 hit me up on Telegram 👊

shandongtlb, 2022-04-17:
> Definitely! I also have the 3. I didn't have time to bindiff the first but both should be relatively easy. I have only little time so I postponed but I have some time here snd there 😃 Both chipsets are supported and both arm. Imagine the Hijacker app on your watch 🤯 hit me up on Telegram 👊

OK, it's a great honor

shandongtlb, 2022-04-17:
> Definitely! I also have the 3. I didn't have time to bindiff the first but both should be relatively easy. I have only little time so I postponed but I have some time here snd there 😃 Both chipsets are supported and both arm. Imagine the Hijacker app on your watch 🤯 hit me up on Telegram 👊

Busy with work, I have less free time. Come on! 😃

lasyka, 2022-04-27:
> Support for bcm43436b0 (brcmfmac) has been added recently. If you're familiar with the tools I'd try to disassemble and bindiff brcmfmac/bcmdhd firmwares

This is the bindiff out put info:
Could not find basic block: 00000004
 .....
 .....
Could not find basic block: 00058B76
Setup: 0.16s
primary:   fw_bcmdhd: 1666 functions, 4891 calls
secondary: brcmfmac43436-sdio: 1595 functions, 4828 calls
Matching: 0.15s
matched: 1537 of 1666/1595 (primary/secondary, 1665/1594 non-library)
call graph MD index: primary   74.448
                     secondary 70.1589
Similarity: 88.8388% (Confidence: 99.0592%)

What is next step to make the firmware patch?

DrSchottky, 2022-04-27:
> What is next step to do and How to make this firmware patch?
> 
Load them into your favorite disassembler and try to figure out how the offsets in definitions.mk need to be changed.

yesimxev, 2022-04-28:
@lasyka I'll hop on to the Ticwatch Pro 3 fw then. Please let me know how you progress

yesimxev, 2022-04-28:
Hmm the TWP3 has two firmwares, and the first one is actually the exact same version as the first watch, 9.88.0.0

```
fw_bcm43436b0.bin
fw_bcm43438a1.bin    
```

lasyka, 2022-05-05:
> Hmm the TWP3 has two firmwares, and the first one is actually the exact same version as the TWP, 9.88.0.0. Wondering which one is being used
> 
> ```
> fw_bcm43436b0.bin - 9.88.0.0
> fw_bcm43438a1.bin - 7.45.96.79
> ```

I'm newbie, don't konow how to do next step how to patch the ticwatch wifi firmware.  :(

Ma5onic, 2022-07-14:
> Hmm the TWP3 has two firmwares, and the first one is actually the exact same version as the TWP, 9.88.0.0. Wondering which one is being used
> 
> ```
> fw_bcm43436b0.bin - 9.88.0.0
> fw_bcm43438a1.bin - 7.45.96.79
> ```
@yesimxev 
I did an `adb shell dumpsys wifi` on my TicWatch Pro 3 Ultra GPS (Rubyfish) and here is the relevant info about which is being used:
```
Chipset information :-----------------------------------------------
FW Version is: Firmware: wl0: Jan 14 2021 10:53:53 version 7.45.96.79 (ce0e3d8@SYNA) (r745790) FWID 01-667de1ce es7
  CLM: 7.11.15 (2014-05-26 10:53:55) 
  Chip: a9a6 Rev 1
Driver Version is: Dongle Host Driver, version 100.10.545.2 (r826445-20190806-3)
Supported Feature set: -1
```
Since it loaded `version 7.45.96.79` I think it is safe to assume that fw_bcm43438a1.bin is being used.

yesimxev, 2022-07-14:
Hmm it's a third version. Can't wait to continue

Ma5onic, 2022-07-16:
@yesimxev the nexmon README says:
`bcm43430a1 was wrongly labeled bcm43438 in the past.`
Is it possible that this is actually a bcm43430a1 chip using version firmware 7.45.96.79?

yesimxev, 2022-07-16:
I'll check when there's time. Both needs patches anyway 👌

Ma5onic, 2022-07-16:
I know, I am trying to do that, but I'm new to patching, seems like my assumption is correct 
[image]

yesimxev, 2022-11-10:
> > What is next step to do and How to make this firmware patch?
> 
> Load them into your favorite disassembler and try to figure out how the offsets in definitions.mk need to be changed. @DrSchottky 

I'm going to disassemble tonight. Is it a simple "find and replace" the addresses, according to the definitions.mk from the same chipset? I may be missing something in there. Coukd.you give me just one example say the first thing you'd change, just so I knoe ehst you mean. Doesn't have to be the solution, just in theory. Also, is the RAM bin also needed or is the ROM fw bin enough? I also have an lg v20 waiting in the nexmon cue, again it's a supported chipset 👌😄

yesimxev, 2022-11-10:
For example: I see 
```
# original ucode start and size
UCODESTART=0x4E9C0
``` 
in firmwares/bcm43438/7_45_41_26/definitions.mk
I'm looking at the disassembled brcmfmac43430-sdio.bin in ghidra. Looks like I found the ucode start in fw_bcm43438a1.bin.
[image]

yesimxev, 2022-11-11:
Almost done boardconfig.mk. What's the next step?

matthiasseemoo, 2022-11-11:
Make a very simple patch that is doing nothing else then extracting the compressed ucode. If that is working and the chip starts your patched firmware, you know that you found the correct locations of the ucode and the minimum required set of functions required for the ucode decompression operation. Then you can continue to add additional patches one by one, such as monitor mode, frame injection, ...

yesimxev, 2022-11-13:
> Make a very simple patch that is doing nothing else then extracting the compressed ucode. If that is working and the chip starts your patched firmware, you know that you found the correct locations [...]

Do you mean just to use/edit the makefile which extracts the ucode? Then use that ucode.bin as fw?

yesimxev, 2022-11-17:
In the process, I was about to have a look at Nexus 6P, see if there's anything wrong in the definitions.mk. (it has an issue, EAPOL packets can't be found in captured handshake) Please advise, am I looking at it in a wrong angle, or does these addresses seem out of memory? Example HNDRATE_RECLAIM_END=0x21ac70 whereas the firmware finishes in 0x9adf8. Also, why does the definitions mention rom.bin, am I missing that?
Thanks for your time.

[image]

DrSchottky, 2022-11-20:
Setup the folders/files structure for a new fw with only the patched for ucode extraction (see patch.c) and try to build and run it

yesimxev, 2022-12-19:
I'm not too sure and sorry if I ask too much. I feel like I won't progress if I don't do so. Does only this patch line to be in patch.c?
[image]

yesimxev, 2022-12-20:
Seems like it's built fw_bcmdhd.bin and it runs on the device. I just want to make sure it's done the ucode extraction. The output was as follows:
[image]

Ma5onic, 2023-04-13:
@yesimxev I noticed that the Raspberry Pi Pico Wireless uses a [Infineon CYW43439](https://www.infineon.com/cms/en/product/wireless-connectivity/airoc-wi-fi-plus-bluetooth-combos/cyw43439/) which has the same architecture on the WLAN side.
[image]
[image]

yesimxev, 2023-04-13:
Thank you for your advice. I also have a big support from @jlinktu, I just have some other stuff to clear first. We're not far from the result 🎉

decaduto, 2023-07-10:
> Thank you for your advice. I also have a big support from @jlinktu, I just have some other stuff to clear first. We're not far from the result tada

any progress? I am starting to porting the work too for the same ticwatch model, let me know if I must work on it or if you have already modded nexmon for that purpose.

yesimxev, 2023-07-10:
It's very close to finish. Just figuring out dunping rom and it doesn't want to spit out what we want. Does this work on your watch in adb shell? `dhdutil -i wlan0 membytes -r 0x181000 0x915ac > /sdcard/rom.bin` (invalid arg on mine)

decaduto, 2023-07-10:
> `dhdutil -i wlan0 membytes -r 0x181000 0x915ac > /sdcard/rom.bin`

already tried to do that sometime ago, same result, even with twrp and root, it fails.

decaduto, 2023-07-10:
> It's very close to finish. Just figuring out dunping rom and it doesn't want to spit out what we want. Does this work on your watch in adb shell? `dhdutil -i wlan0 membytes -r 0x181000 0x915ac > /sdcard/rom.bin` (invalid arg on mine)

pheraps, have you found the offset of some "Important" function? (monitor mode, send frame, rx handler ecc..)

yesimxev, 2023-09-29:
I want to first copy the ROM somewhere into a RAM location and then dump it. But I'm not sure how to access

yesimxev, 2024-02-15:
> > `dhdutil -i wlan0 membytes -r 0x181000 0x915ac > /sdcard/rom.bin`
> 
> already tried to do that sometime ago, same result, even with twrp and root, it fails.

I tried the other method which I see on this repo for other devices with my address. But it returns nex_driver_io error. There must be a step in between that lets me access that memory address

[image]

yesimxev, 2024-02-16:
Thanks for your reply, I tried that too, but I had only this screenshot at hand. I'm still about to do the summary where are we at 😃

yesimxev, 2024-02-16:
I came across this readme in one of the rom-extraction folders, saying 

```
The **ROM extraction patch adds additional ioctls** to the Wi-Fi firmware to dump arbitrary memory locations. To dump the first 1024 bytes of the ROM, one may execute: `nexutil -g0x602 -l1024 -i -v0x0 -r > /var/root/romdump.bin`. As the buffer length for each ioctl is limited, we need to call the ioctl multiple times with different start addresses passed through `-v<address>`. The resulting ROM ...
```

So I'm trying to understand. Do I need to copy the patched fw back to the device? As far as I know, my patches which were done only did the extracted ucode part. Is there a step missing before I could run dump-rom?