seemoo-lab/nexmon #519: Samsung Galaxy S20 G980F Monitor Mode Fails

agrossmann09, 2022-03-17:
I currently have my S20 G980F setup as follows:
- Android 11 / OneUI 3.1
- ThundeRStormS OneUI 3.1 Kernel (Magisk is installed when flashing this kernel)
- TWRP
- The original wlan firmware version was: 18.41.104.4
- Placing the new patched firmware under vendor/etc/wifi did not work. I have to place it under /system/etc/firmware. After doing so my new firmware is 18.41.8.9 which matches the nexmon firmware. 
- I have SELinux disabled on boot using Magisk. I have also disabled it through the command line or by using ThunderTweaks. 

Below I have included some output logs.

```
[ubuntu@192.168.0.25] ~:$ adb shell
* daemon not running; starting now at tcp:5037
* daemon started successfully
x1s:/ $ su
x1s:/ # getenforce
Permissive
x1s:/ # nexutil -V
platform SM-G980F
firmware 18.41.8.9 (B1 Network/rsdb) FWID 01-41d34ca2
vendorid 0x14e4
deviceid 0x4470
radiorev 0xa05fb
chipnum 0x4375
chiprev 0x5
chippackage 0x0
corerev 0x52
boardid 0x88f
boardvendor 0x14e4
boardrev P300
driverrev 0x0
ucoderev 0x5a02710
bus 0x0
phytype 0xb
phyrev 0x2c
anarev 0x0
nvramrev 0xc202a

platform             | firmware                         | vendorid | deviceid | radiorev   | chipnum | chiprev | chippackage | corerev | boardid | boardvendor | boardrev | driverrev | ucoderev  | bus | phytype | phyrev | anarev | nvramrev
-------------------- | -------------------------------- | -------- | -------- | ---------- | ------- | ------- | ----------- | ------- | ------- | ----------- | -------- | --------- | --------- | --- | ------- | ------ | ------ | --------
SM-G980F             | 18.41.8.9 (B1 Network/rsdb) FWID 01-41d34ca2 |   0x14e4 |   0x4470 |    0xa05fb |  0x4375 |     0x5 |         0x0 |    0x52 |   0x88f |      0x14e4 |     P300 |       0x0 | 0x5a02710 | 0x0 |     0xb |   0x2c |    0x0 |  0xc202a
x1s:/ # iw wlan0 info
Interface wlan0
	ifindex 17
	wdev 0x1
	addr 6a:e8:a5:c3:08:d8
	type managed
	wiphy 0
x1s:/ # ifconfig
wlan0     Link encap:Ethernet  HWaddr 6a:e8:a5:c3:08:d8
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0 
          TX packets:6 errors:0 dropped:0 overruns:0 carrier:0 
          collisions:0 txqueuelen:1000 
          RX bytes:0 TX bytes:516 

p2p0      Link encap:Ethernet  HWaddr 8e:b8:4a:e1:3c:df
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0 
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0 
          collisions:0 txqueuelen:1000 
          RX bytes:0 TX bytes:0 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0 
          inet6 addr: ::1/128 Scope: Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0 
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0 
          collisions:0 txqueuelen:1000 
          RX bytes:0 TX bytes:0 

x1s:/ # nexutil -m2
x1s:/ # iw wlan0 info
Interface wlan0
	ifindex 17
	wdev 0x1
	addr 6a:e8:a5:c3:08:d8
	type managed
	wiphy 0
x1s:/ # getprop | grep wlan                                                                                                                                                                               
[ro.security.wlan.release]: [2]
[ro.security.wlan.ver]: [1.0]
[ro.telephony.iwlan_operation_mode]: [legacy]
[vendor.wifi.dualconcurrent.interface]: [swlan0]
[vendor.wlan.driver.version]: [101.16.79.6]
[vendor.wlan.firmware.version]: [18.41.8.9]
[wifi.active.interface]: [wlan0]
[wifi.interface]: [wlan0]
[wlan.driver.status]: [ok]
[wlan.p2p.chkintent]: [0]
[wlan.p2p.temp]: [-3]
[wlan.p2p.wfdsta]: [disconnected]
```

In order to capture on wlan0 I would use tcpdump. I also included the libfakeioctl.so in the call to tcpdump. Even if I try to use iw to set the monitor mode I get a -95 error code.

agrossmann09, 2022-03-17:
I do get the following numbers when running nexutils -m
-2147483548
-2147483645
-2147483646

agrossmann09, 2022-03-19:
I am now testing with BeyondROM which also runs ThundeRStormS kernel. But it runs on Android 12 and OneUI 4.0

agrossmann09, 2022-03-21:
I also had to make some changes to the install-firmware script since on the Samsung Galaxy 20 the firmware is not located at vendor/etc/wifi but at vendor/firmware.

pjk2563, 2022-08-09:
@matthiasseemoo 
I'm having a very similar issue and before patching my own firmware I was wondering if you had any insight. 

My differences:
Platform: Galaxy S10 G973F
Kernel: cruelkernel 4.14.113
Stock Firmware: 18.41.75 
Installation: Magisk module to overwrite /vendor/firmware/bcmdhd_sta.bin_b1 on boot 

Running nexutil -V with the stock firmware shows 18.41.75 and ucoderev 0x5a02710 as the only differences between the provided 18_38_18 firmware. 

When I install the firmware, WiFi works as normal and nexutil -V reports the new version, so I'm assuming it's compatible. However, trying to get anything out of monitor mode doesn't work as per @agrossmann09's posts. 

Any ideas what might be the issue?

matthiasseemoo, 2022-10-11:
Normally, firmware versions need to be compatible with the driver in use. My best guess would be to apply the nexmon patches to the new stock firmware and them try it again.

pjk2563, 2024-01-25:
> @KernelAtom <https://github.com/KernelAtom>
> you can get the original wifi firmware by fetching the stock samsung
> android image of the version you are currently running, there are several
> tools out there that allow you to do this.
>
> once you have the stock android image you can find the bcmdhd_sta.bin_b1
> residing in the vendor partition.
>
> here is an example on how to get the wifi firmware based on samsungs
> G980FXXSIHWJD image for the SM-G980F:
>
> $ unzip -l SM-G980F_4_20231101175920_732z45uwem_fac_G980FXXSIHWJD_G980FOXMIHWH9_G980FXXSIHWJD_G980FXXSIHWJD_DBT.zip | grep AP_
> 7797371002  2023-11-01 18:44   AP_G980FXXSIHWJD_G980FXXSIHWJD_MQB73111678_REV01_user_low_ship_MULTI_CERT_meta_OS13.tar.md5
> $ unzip SM-G980F_4_20231101175920_732z45uwem_fac_G980FXXSIHWJD_G980FOXMIHWH9_G980FXXSIHWJD_G980FXXSIHWJD_DBT.zip AP_G980FXXSIHWJD_G980FXXSIHWJD_MQB73111678_REV01_user_low_ship_MULTI_CERT_meta_OS13.tar.md5
> Archive:  SM-G980F_4_20231101175920_732z45uwem_fac_G980FXXSIHWJD_G980FOXMIHWH9_G980FXXSIHWJD_G980FXXSIHWJD_DBT.zip
>   inflating: AP_G980FXXSIHWJD_G980FXXSIHWJD_MQB73111678_REV01_user_low_ship_MULTI_CERT_meta_OS13.tar.md5
> $ tar -xf AP_G980FXXSIHWJD_G980FXXSIHWJD_MQB73111678_REV01_user_low_ship_MULTI_CERT_meta_OS13.tar.md5 super.img.lz4
> $ unlz4 super.img.lz4
> Decoding file super.img
> super.img.lz4        : decoded 8946059092 bytes
> $ simg2img super.img super.img.raw
> $ lpunpack -p vendor super.img.raw
> Attempting to extract partition 'vendor'...
>   Dealing with extent 0 from target source 0...
> $ mount -o loop vendor.img /mnt/loop
> $ strings /mnt/loop/firmware/bcmdhd_sta.bin_b1 | tail -n1
> <FW-TAG>4375b1-roml/config_pcie_release Version=18.41.117 (B1 Network/rsdb) Date=2022-07-19T02:10:11Z FWID=01-1b4772fd
>
> note that in this case the original wifi firmware is of a newer release (
> 18.41.117) than the one you are searching for.. thus you want to make
> sure to fetch the samsung firmware you are currently running on your phone
> instead of the version I used in the example

So once we have the firmware, how would we go about patching it?

hjchjcjh, 2024-04-03:
I have the same problem too. My phone is g981n. with Android 13 / OneUI 3
When I put in a lot of effort to make selinux become permissive mode. Unfortunately, nexmon seem no work  in my phone