seemoo-lab/nexmon #524: Build for Samsung Galaxy Note 3 Neo (SM-7505) running LineageOS 14.1

tranzmatt, 2022-04-27:
I wanted to run both hijacker (Airodump-ng) and networksurvey on my SM-7505, but networksurvey requires Android 7+, so I upgraded from Android 5.1.1 to Lineage 14.1.  It installed fine, the device is rooted, but now I can't get the wifi into monitor mode for Airodump-ng and the nexmon app fails to install the modded version, which I guess makes sense since it's expecting Android 5.  How can I build/install a version of the bcm4339 driver for it?  I'm willing to read directions, I'm just not familiar with this particular environment.

tranzmatt, 2022-04-28:
For that matter, is the BCM4339 even the right chipset?  I'm seeing conflicting information about whether it's a 4339 or 4335.

matthiasseemoo, 2022-04-28:
run nexutil -V to check the version information. You can compare it to the entries in https://github.com/seemoo-lab/nexmon/blob/master/REVINFO.md

tranzmatt, 2022-04-29:
> run nexutil -V to check the version information. You can compare it to the entries in https://github.com/seemoo-lab/nexmon/blob/master/REVINFO.md
> […](#)

This is what I got back.  It looks like I should be able to leverage the Nexus 5 version, but I'm not sure what file to replace.  I added the results of some greps through dmesg:

```
platform SM-N7505
firmware 6.37.32.47 (A0 Station/P2P)
vendorid 0x14e4
deviceid 0x43ae
radiorev 0x92069000
chipnum 0x4339
chiprev 0x1
chippackage 0x2
corerev 0x2e
boardid 0x6b5
boardvendor 0x14e4
boardrev P100
driverrev 0x62520
ucoderev 0x35728a6
bus 0x0
phytype 0xb
phyrev 0x6
anarev 0x0
nvramrev 0x0

$ grep 4339 ~/Android/neo3-dmesg.txt
[    8.174339] [4]      1 max77803: c: 0xb5(0x00)
[   11.556062] [4]      1 F1 signature OK, socitype:0x1 chip:0x4339 rev:0x1 pkg:0x2
[   11.611079] [4]      1 Compiled in drivers/net/wireless/bcmdhd4339 on Dec 31 2020 at 15:07:53
[   26.640712] [3]   2690 Compiled in drivers/net/wireless/bcmdhd4339 on Dec 31 2020 at 15:07:53
[   27.204595] [2]   2690 F1 signature OK, socitype:0x1 chip:0x4339 rev:0x1 pkg:0x2
$ grep firmware ~/Android/neo3-dmesg.txt
[    5.525717] [4]      1 fts 5-0049: fts_fw_update_on_probe: Load firmware : tsp_stm/stm_de.fw
[   27.228192] [0]   2690 dhdsdio_download_firmware: firmware path=/system/etc/wifi/bcmdhd_sta.bin, nvram path=/system/etc/wifi/nvram_net.txt
```

matthiasseemoo, 2022-04-29:
You need to replace /system/etc/wifi/bcmdhd_sta.bin

> On 29. Apr 2022, at 05:55, Matthew Clark ***@***.***> wrote:
> 
> /system/etc/wifi/bcmdhd_sta.bin

tranzmatt, 2022-04-30:
Sadly, just replacing bcmdhd_sta.bin with fw_bcm4339_6_37_34_43.bin built by nexmon broke things

```
[   26.190444] [2]   2688 Compiled in drivers/net/wireless/bcmdhd4339 on Dec 31 2020 at 15:07:53
[   26.632842] [0]   2688 dhdsdio_download_firmware: firmware path=/system/etc/wifi/bcmdhd_sta.bin, nvram path=/system/etc/wifi/nvram_net.txt
[   26.987524] [4]   2688 dhdsdio_write_vars: Download, Upload and compare of NVRAM succeeded.
[   36.141231] [4]   2688 dhd_bus_init: enable 0x06, ready 0x02 (waited 3003191us)
[   41.141605] [1]   2688 dhd_bus_rxctl: resumed on timeout, INT status=0xA0000080
[   41.143683] [1]   2688 dhd_bus_rxctl: rxcnt_timeout=1, rxlen=0
[   41.143734] [1]   2688 dhd_check_hang: Event HANG send up due to  re=1 te=0 e=-110 s=1
[   41.143822] [1]   2688 dhd_wl_ioctl: WLC_SET_VAR: cur_etheraddr, ret = -110
[   41.143907] [1]   2688 [WIFI_SEC] dhd_bus_start: _dhd_set_mac_address() failed
[   41.143955] [1]   2688 dhd_prot_ioctl : bus is down. we have nothing to do
[   41.144000] [1]   2688 dhd_wl_ioctl: WLC_IOCTL: cmd: 98, ret = -1
[   41.144043] [1]   2688 dhd_open : wl_android_wifi_on failed (-14)
[   41.144098] [1]   2688 dhd_prot_ioctl : bus is down. we have nothing to do
[   41.144140] [1]   2688 dhd_wl_ioctl: WLC_GET_VAR: bss, ret = -1
[   41.144339] [1]   2688 dhd_prot_ioctl : bus is down. we have nothing to do
[   41.144389] [1]   2688 dhd_wl_ioctl: WLC_IOCTL: cmd: 3, ret = -1
```

matthiasseemoo, 2022-05-01:
Then you have to port the patches to the new firmware version.

tranzmatt, 2022-05-01:
> Then you have to port the patches to the new firmware version.
> […](#)

Is there a guide on this?  I don't mind doing it, but I've never done anything like this before.

matthiasseemoo, 2022-05-01:
> … <x-msg://42/#>

Not really. You need to find the locations of the functions and data required by the patch in the new firmware.
Then, you need to create a definitions.mk file in a subdirectory of firmwares containing some of those addresses generally required for patching.
You need to extend the wrapper.c file with the addresses of functions in the new firmware.
And then you can copy the firmware patch of the old firmware to a new subdurectory under patches.