seemoo-lab/nexmon #526: Pi zero 2 W no working

R2Sam, 2022-05-02:
Having issues with getting any results from using this patch, as a relative linux noob it is simply proving a bit beyond me to try and remedy this myself. Using 5.10 Raspberry OS and followed all instructions in the main readme, with the additional help of #500 I retried many times yet with no success. I simply am not getting any SSID coming up when running airmon-ng with either mon0 or by even using airmon-ng start wlan0 and then using wlan0mon, although interestingly I am getting error  524 when running the later method. Will be happy to share more information and details as requested.

Aiaid, 2022-05-11:
Hi, I got the same problem as you, and I couldn't figure it out either. I found Dr. Schottky uploaded an image for pwnagotchi, which has a working driver for pi zero 2w(https://github.com/evilsocket/pwnagotchi/issues/1046#issuecomment-1059776337).

I copy the driver bin and brcmfmac.ko to a  5.10.63 Raspberry lite OS(https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-11-08/), which work.

https://anonfiles.com/D295qff3yd/rpi02w_zip

```
cp brcmfmac.ko  /lib/modules/5.10.63-v7+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko
cp -r brcm/* /lib/firmware/brcm/
reboot
```
and start mon0 by `iw phy "$(iw phy | head -1 | cut -d" " -f2)" interface add mon0 type monitor && ifconfig mon0 up`

closed 2022-06-16 by R2Sam

favdistro, 2025-02-04:
@Aiaid https://anonfiles.com/D295qff3yd/rpi02w_zip this link not working